scmanjarrez/OctoPrint-SmartABL #1: [Request] - allow for custom G29 codes to be used

mdeeter, 2022-12-30:
### Is your feature request related to a problem? Please describe.

I like to use a custom leveler command in my START code:

G29 X4 Y4 L50 R150 F50 B150 T V4  ; do a quick BED LEVLING : 5 - point matrix bed level check

It would be nice if I could have SmartABL allow for custom input of the G29 command instead of using the default full G29.

### Describe the solution you'd like

I could see two different solutions:

1. I include the custom G29 command in my gcode... so it would be nice if it was just used when doing the leveling... and just skip it until the plugin settings are met (every 3 prints, or +1 days is what I set mine to be)
2. allow the user to define it in the Octoprint plugin ... this seems like more work, but if this was easier, it'd be an okay solution

scmanjarrez, 2022-12-30:
I didn't even know that was possible, I'll work on that, really easy to implement 👍

scmanjarrez, 2022-12-30:
I've released a new version that uses the abl command read from your file. I added a new setting that allow sending custom abl instead if preferred (disabled by default).

mdeeter, 2022-12-30:
Sweet... I'll check it out and test it asap.  Thanks again!

mdeeter, 2022-12-30:
I downloaded the update...

And I made 6 test prints and it never triggered a leveling.

I have this command in my Start G-code in Prusa Slicer: `G29 X4 Y4 L50 R150 F50 B150 T V4` 
(in amongst other commands to prepare the temps and purge)

...but a leveling doesn't occur at all.

I then added `G29 X4 Y4 L50 R150 F50 B150 T V4` to the custom input and tried a test print... and it still didn't auto-level.

[image]

I turned off the plugin and ran another test print... and the G29 command (above) triggered fine.

I'm running a Creality Ender 3 S1
Octoprint v1.8.6

Here's my full start code:
```

M140 S{first_layer_bed_temperature[0]} ; SET BED TEMP to first layer temp

G90 ; Absolute positioning

M220 S100 ; Reset Feedrate
M221 S100 ; Reset Flowrate

G28 ; Home all axes

; setting this in increments allows for a faster cancel if needed... 
; otherwise, you'll have to wait for the bed to fully heat up if you want to cancel
M190 S{first_layer_bed_temperature[0]*.3} ; Wait for Heat Bed temp to be 30%
M190 S{first_layer_bed_temperature[0]*.4} ; Wait for Heat Bed temp to be 40%
M190 S{first_layer_bed_temperature[0]*.5} ; Wait for Heat Bed temp to be 50%
M190 S{first_layer_bed_temperature[0]*.6} ; Wait for Heat Bed temp to be 60%
M190 S{first_layer_bed_temperature[0]*.7} ; Wait for Heat Bed temp to be 70%
M190 S{first_layer_bed_temperature[0]*.8} ; Wait for Heat Bed temp to be 80%

; bed is almost warmed up... set extruder temp to 80% of first layer temp to get it ready
M104 S{first_layer_temperature[0]*.8} ; start warming extruder to partial full temp - not full so it doesn't drip during next steps

M190 S{first_layer_bed_temperature[0]*.9} ; Wait for Heat Bed temp to be 90%

M140 S{first_layer_bed_temperature[0]} ; Set 100% Bed temp so it keeps going wile leveling

; BED LEVELING
; do a quick BED LEVLING : 5 - point matrix bed level check
G29 X4 Y4 L50 R150 F50 B150 T V4  

M500 ; save bed leveling to system

G1 X5 Y-3.5 Z40 F5000.0 E-0.5 ; move to ooze position and retract a little bit

G92 E0 ; zero the extruded length

M104 S{first_layer_temperature[0]} ; Set heating extruder to 100% first layer print temp

M190 S{first_layer_bed_temperature[0]} ; Wait for Heat Bed temp to be 100%

M109 S{first_layer_temperature[0]} ; Wait for Extruder temp to be 100% for first layer

; START PURGE PROCESS

M106 S76.5 ; turn fan on

G92 E0 ; reset extruder

; BLOB 1
G0 X1 Y5 Z0.8 F3000.0 ; move to blob 1 position
G1 F200 E7 ; extrude blob 1
G1 Z5 E11; move the extruder up by 5mm while extruding more
G0 Z10.00 F10000; move up a bit more
G92 E0 ; reset extruder

; BLOB 2
G0 X1 Y10 Z0.8 F3000.0 ; move to blob 2 position
G1 F200 E7 ;extrude blob 2
G1 Z5 E11; move the extruder up by 5mm while extruding more
G0 X1 Y16 Z.08 F10000; move away from blob 2
G92 E0 ; reset extruder

; FINAL WIPE
G0 X1 Y16 Z0.3 F5000; wipe 
G1 X1 Y22 E.5 Z0.3 F5000 ; wipe 
G92 E0;

G0 X1 Y50 Z20 F10000 ; lift Z so blobs aren't touched by CR Touch
G0 X120 Y120 F5000; moe to the center-ish
```

scmanjarrez, 2022-12-30:
Do you mind enabling logging in settings, doing a print and then sharing the file plugin_SmartABL.log?
[image]

mdeeter, 2022-12-30:
log data from `plugin_SmartABL.log`

```2022-12-30 17:30:13,861 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintStarted)
2022-12-30 17:30:13,864 @on_event:print_start >> Mesh query
2022-12-30 17:30:29,624 @process_line:mesh_valid >> valid_mesh=True
2022-12-30 17:30:29,959 @queuing_gcode > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True)
2022-12-30 17:30:29,963 @queuing_gcode:abl_skip >> Sending M420 S1
2022-12-30 17:30:29,982 @sent_gcode > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(gcode=M500)
2022-12-30 17:30:29,984 @sent_gcode:mesh_update >> prints=0, last_mesh=30/12/2022
2022-12-30 17:36:13,697 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=0, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintFailed)
2022-12-30 17:36:13,699 @on_event:print_stop >> prints=0
```

scmanjarrez, 2022-12-30:
Oh, I see the problem, you have M500 in your gcode, so the counter is reset to 0 everytime it reads it after loading the mesh. I'll try to take into account that case, meanwhile, do you mind commenting M500 from your starting script?

mdeeter, 2022-12-30:
Sure np.  Will comment out and retest.

I'm not a gcode genius (only been printing since Sept)... my understanding is that after the G29 command, one would want to save those leveling results using the M500.  Does that seem like an accurate understanding?

scmanjarrez, 2022-12-30:
That's correct, but I'm doing that in the plugin, when it reads G29*:
- If the mesh is outdated, it sends G29 and M500 (this last one to save to eeprom)
- If the mesh is updated, it rewrites it to M420 S1 (load mesh from eeprom)

When the plugin reads M500 (assuming I'm the one sending it) it resets the counter of "prints". After a successfuly print, this counter is incremented.

In a "normal" case, the plugin sees G29*, load the mesh from eeprom and after the print finishes, increment the counter. In this case it reads another M500 (not expected) and the counter is reset, so... it never reaches the condition value.

mdeeter, 2022-12-31:
Yup, that makes sense.  I didn't realize that the M500 was affecting the count.

I commented out the M500... and no luck.  I started it several times and cancelled the print after I saw it go to the Ooze position, which means it skipped the G29 code in my start code:

```
M140 S{first_layer_bed_temperature[0]} ; SET BED TEMP to first layer temp

G90 ; Absolute positioning

M220 S100 ; Reset Feedrate
M221 S100 ; Reset Flowrate

G28 ; Home all axes

; setting this in increments allows for a faster cancel if needed... 
; otherwise, you'll have to wait for the bed to fully heat up if you want to cancel
M190 S{first_layer_bed_temperature[0]*.3} ; Wait for Heat Bed temp to be 30%
M190 S{first_layer_bed_temperature[0]*.4} ; Wait for Heat Bed temp to be 40%
M190 S{first_layer_bed_temperature[0]*.5} ; Wait for Heat Bed temp to be 50%
M190 S{first_layer_bed_temperature[0]*.6} ; Wait for Heat Bed temp to be 60%
M190 S{first_layer_bed_temperature[0]*.7} ; Wait for Heat Bed temp to be 70%
M190 S{first_layer_bed_temperature[0]*.8} ; Wait for Heat Bed temp to be 80%

; bed is almost warmed up... set extruder temp to 80% of first layer temp to get it ready
M104 S{first_layer_temperature[0]*.8} ; start warming extruder to partial full temp - not full so it doesn't drip during next steps

M190 S{first_layer_bed_temperature[0]*.9} ; Wait for Heat Bed temp to be 90%

M140 S{first_layer_bed_temperature[0]} ; Set 100% Bed temp so it keeps going wile leveling

; BED LEVELING
; do a quick BED LEVLING : 5 - point matrix bed level check
G29 X4 Y4 L50 R150 F50 B150 T V4  

; M500 ; save bed leveling to system

G0 X5 Y-3.5 Z40 F5000.0 ; move to ooze position.     ; <!---- I cancelled each print attempt when I saw it get to here

G92 E0 ; zero the extruded length
```

Here's the logs after several starts:
```
2022-12-30 17:30:13,861 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintStarted)
2022-12-30 17:30:13,864 @on_event:print_start >> Mesh query
2022-12-30 17:30:29,624 @process_line:mesh_valid >> valid_mesh=True
2022-12-30 17:30:29,959 @queuing_gcode > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True)
2022-12-30 17:30:29,963 @queuing_gcode:abl_skip >> Sending M420 S1
2022-12-30 17:30:29,982 @sent_gcode > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(gcode=M500)
2022-12-30 17:30:29,984 @sent_gcode:mesh_update >> prints=0, last_mesh=30/12/2022
2022-12-30 17:36:13,697 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=0, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintFailed)
2022-12-30 17:36:13,699 @on_event:print_stop >> prints=0
2022-12-30 18:05:33,232 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintStarted)
2022-12-30 18:05:33,241 @on_event:print_start >> Mesh query
2022-12-30 18:05:50,266 @process_line:mesh_valid >> valid_mesh=True
2022-12-30 18:05:50,599 @queuing_gcode > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True)
2022-12-30 18:05:50,604 @queuing_gcode:abl_skip >> Sending M420 S1
2022-12-30 18:06:37,149 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintFailed)
2022-12-30 18:06:37,150 @on_event:print_stop >> prints=1
2022-12-30 18:09:14,484 @initialize > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=False)
2022-12-30 18:09:46,300 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=False) || Trigger(event=PrintStarted)
2022-12-30 18:09:46,307 @on_event:print_start >> Mesh query
2022-12-30 18:10:01,986 @process_line:mesh_valid >> valid_mesh=True
2022-12-30 18:10:02,333 @queuing_gcode > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True)
2022-12-30 18:10:02,340 @queuing_gcode:abl_skip >> Sending M420 S1
2022-12-30 18:10:45,597 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintFailed)
2022-12-30 18:10:45,602 @on_event:print_stop >> prints=1
2022-12-30 18:11:45,284 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintStarted)
2022-12-30 18:11:45,287 @on_event:print_start >> Mesh query
2022-12-30 18:12:04,596 @process_line:mesh_valid >> valid_mesh=True
[... cut ...]
2022-12-30 18:14:13,161 @queuing_gcode > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True)
2022-12-30 18:14:13,166 @queuing_gcode:abl_skip >> Sending M420 S1
2022-12-30 18:14:20,264 @on_event > Settings(force_days=1, days=1, force_prints=1, prints=3), failed=0) || State(first_time=False, prints=1, last_mesh=30/12/2022) || Internal(valid_mesh=True) || Trigger(event=PrintFailed)
2022-12-30 18:14:20,274 @on_event:print_stop >> prints=1
```

scmanjarrez, 2022-12-31:
Well, this time it isn't working because you didn't complete a print. If you want to check that it really works, enable "PrintFailed" in the settings. 
The counter isn't incrementing without a successful print.

mdeeter, 2022-12-31:
ahh.... man, I must be getting tired.  I noticed that feature in the plugin settings, but forgot about that impact.  Retrying with the obviously well-documented setting turned.  Sorry for the booger-brain.

mdeeter, 2022-12-31:
Success!

Thanks a ton for adding the ability to use the custom G29.

I was starting to research how to make this possible just the other day... so glad I saw the "New plugin" announcement... this was one of the last hurdles I wanted to help keep things running smoothly.  I've massaged that start code quite a bit and I really like the affect - and now this plugin helps it not feel so overwhelming.

Enjoy your weekend!  Feel free to close this issue at your leisure.

scmanjarrez, 2022-12-31:
Thank you! And happy new year :)

And yes, I created this plugin for the exact same reason, I'm glad someone find it useful too 😄

scmanjarrez, 2022-12-31:
I'll keep it open until I release the new version handling the M500 problem (already fixed, but I'll add another feature to disable smartabl on the fly, without going to settings; and forcing abl for just one print)

scmanjarrez, 2022-12-31:
I have release a pre-release version with a new feature. This bug is already fixed in https://github.com/scmanjarrez/OctoPrint-SmartABL/tree/0.1.3rc0. Let me know if you find any error with your start gcode.

scmanjarrez, 2023-01-02:
Closed as implemented in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.3

closed 2023-01-02 by scmanjarrez