scmanjarrez/OctoPrint-SmartABL #2: [Request] - Log previous print bed temperature and do ABL if changed

elpeter25, 2022-12-31:
### Is your feature request related to a problem? Please describe.

Hi,
First thanks for this plugin, it's awesome!

I had the idea of avoid doing ABL if let's say previous print or the print where the ABL was trigered was done with same bed temperature but do it if it changed.

i.e.: Previous 3 prints done with PLA at 60º bed and ABL done on first print, next print going for PETG at 80º bed so perform ABL.

P.S.: Maybe instead of writing a complex way to save history of previous prints take data from [PrintJobHistory](https://github.com/OllisGit/OctoPrint-PrintJobHistory) plug-in

Thanks in advance!!

### Describe the solution you'd like

Previous print bed temperature differs form next one, so perform ABL.

scmanjarrez, 2022-12-31:
You want to trigger ABL if the current bed temperature is different than the last one? Is that correct?
> Btw, I have released a pre-release with some buttons in the side panel to have better control (manual) on the triggered ABL

scmanjarrez, 2023-01-01:
I've created a new pre-release with this feature, do you mind checking if everything is working as described in your request? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.3rc1

mdeeter, 2023-01-01:
Love these changes... so cool of you to oblige us.

[image]

scmanjarrez, 2023-01-02:
Closed as implemented in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.3

closed 2023-01-02 by scmanjarrez

elpeter25, 2023-01-03:
That is exactly what I was talking about. Thanks so much!!