scmanjarrez/OctoPrint-SmartABL #4: [Help needed] Looking for Klipper users to make the plugin compatible

scmanjarrez, 2023-01-03:
Hi,
I'm looking for Klippers users that want to help me testing the plugin.
I'd like to understand how it works, how the leveling process is done with Klipper. 
- Do you use the same gcodes (G29*)? 
- How to you generate the gcodes files (same slicers a marlin)? A copy of gcodes file would be very helpful. 
- Is it possible to save and load a mesh from eeprom/disk?

iceparr0t, 2023-01-07:
@scmanjarrez Klipper user who would be willing to help get SmartABL compatible with Klipper native commands and test it out.   Klipper has a few of the basic Marlin Gcode G/M commands ported to make things easier for users switching from Marlin to Klipper, but the ABL sequence does not use G29 but rather direct Klipper commands.

You can find the documentation here (https://www.klipper3d.org/Bed_Mesh.html) but I've given my thoughts below on how to implement Klipper support since it's just a couple of commands replacing what you are using in Marlin. 

  Main commands are:

BED_MESH_CALIBRATE      

This command is the equivalent of G29 in that it will cause ABL to run and save the resulting mesh to the "default" profile.

(Note that all Klipper commands *are* case sensitive, including profile names)

I'd recommend instead using a specific profile for SmartABL, such as "SmartABL" using the command

BED_MESH_CALIBRATE PROFILE=SmartABL

To load a bed mesh one uses the command

BED_MESH_PROFILE LOAD=<name>

Therefore for the above example of saving the mesh to the "SmartABL" profile, the command would be

BED_MESH_PROFILE LOAD=SmartABL

Klipper will either respond with 

Send: BED_MESH_PROFILE LOAD=SmartABL
Recv: ok

or 

Recv: !! bed_mesh: Unknown profile [SmartABL]
Recv: ok

if no such profile exists, which then you could use to trigger the need to run the calibration.

As an example of starting Klipper Gcode that Octoprint would need to parse - you'd be looking for the BED_MESH_CALIBRATE command.

G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S{first_layer_temperature[0]*0.5} ; set extruder temp to 50% of usable while heating bed
M190 S[first_layer_bed_temperature] ; set bed temp
G28 ; home all (once bed is hot)
BED_MESH_CALIBRATE; level bed (once bed is hot)

scmanjarrez, 2023-01-08:
Thank you very much!! I read the documentation few days ago and some notes from a user in another octoprint plugin, but wasn't really sure how it worked. This information is really useful. I'll try to implement this for version 0.1.4

scmanjarrez, 2023-01-08:
Do you mind sharing the message (terminal) that you receive after connecting the printer with octoprint?

iceparr0t, 2023-01-13:
I'm using OctoKlipper (which I expect all Octoprint + Klipper users are using) - this is the message I get in terminal after connecting to Klipper.

`Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /tmp/printer@115200, /tmp/printer@250000, /tmp/printer@230400, /tmp/printer@57600, /tmp/printer@38400, /tmp/printer@19200, /tmp/printer@9600
Trying port /tmp/printer, baudrate 115200
Connecting to port /tmp/printer, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xb1bf9250, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.9.1-617-gf2421005 FIRMWARE_NAME:Klipper
Send: M21
Recv: ok`

jaybz, 2023-01-18:
Just want to point out a possible pitfall with Klipper compatibility:

Klipper saves bed mesh profiles in-memory of the RPi (or whatever you're using to run Octoprint) until a `SAVE_CONFIG` GCode is sent. `SAVE_CONFIG`, however, also saves other in-memory config changes (eg, z offset) and then restarts Klipper. `SAVE_CONFIG` does not disconnect OctoPrint, but it will likely interrupt ongoing prints. Also, if the mesh profile is saved via `SAVE_CONFIG` and then a new mesh is generated using the same profile name but then isn't saved via `SAVE_CONFIG`, restarting Klipper will load the older mesh and the newer one will be discarded.

I suspect SmartABL might not be able to detect that an older profile was loaded without taking additional steps to ensure consistency between the mesh generation date/time that SmartABL saves for mesh age tracking and actual mesh generation date/time. To make things more complicated, issuing a `FIRMWARE_RESTART` GCode restarts Klipper which also reloads saved mesh profiles and discards unsaved ones **_while still keeping Octoprint connected_**.  One way to detect a restart is to wait for Octoprint to receive a "// Klipper state: Ready" from Klipper.

Possible workarounds:
1. Issue a `BED_MESH_PROFILE REMOVE=<profile>` command when Octoprint receives "// Klipper state: Ready" or when Octoprint connects to Klipper, and then proceed as if there was no saved mesh. This does mean that meshes that would have been considered "still good" would be discarded whenever Octoprint has to reconnect.
2. Replace the mesh generation date with the last modified date of printer.cfg (where Klipper saves profiles) when Octoprint connects to Klipper. The actual path to the printer.cfg can be taken from the klippy command line. One caveat here though is that Octoprint might have connected to Klipper after new meshes were generated already. This means that the printer.cfg would be older than the actual mesh profile.
3. You can also use `BED_MESH_OUTPUT` to detect if a bed mesh is active. It will output bed mesh points if one is loaded and an informational message if no mesh is active. Issuing a `BED_MESH_CLEAR` will clear the active bed mesh. `BED_MESH_OUTPUT`, unfortunately will not indicate the date/time when the mesh was generated, but it should possible to checksum the output to detect changes if you use `BED_MESH_OUTPUT PGP=0`. Note that `BED_MESH_OUTPUT PGP=1` only outputs probe points used, not the actual mesh points.

I haven't had time to get possible outputs for all these commands yet, but if you need specific sample outputs, I can get that for you in the meantime.

scmanjarrez, 2023-01-18:
I was just thinking to drop save/load feature from Prusa/Klipper.
- Prusa doesn't have eeprom enabled
- SAVE_CONFIG restarts klipper which would discard the print job.

I can't assume printer.cfg being accesible, there are lot of possibilities (running it inside a container, another user account, etc).
> I have installed klipper myself due to this post, so I've been testing it a lot.

jaybz, 2023-01-18:
Not sure if this will be useful, but you can also save variables in Klipper to disk if enabled via configuration: https://www.klipper3d.org/Command_Templates.html#save-variables-to-disk

jaybz, 2023-01-18:
I've been testing a set of macros for Klipper to implement enough of `G29`, `M500`, and `M420` to make SmartABL work btw. The macros work so far although I've only been testing using a gcode file that does G28 then G29. I've not tested this on actual prints yet. The following are the relevant entries in the cfg file: (Note that this performs a `BED_MESH_CLEAR` the mesh on Klipper start)

[gcode_macro G29]
gcode:
    BED_MESH_CALIBRATE

[gcode_macro M500]
gcode:
    {action_respond_info('SAVE IGNORED!')}

[gcode_macro M420]
gcode:
    {% if printer.bed_mesh.profile_name == ''  %}
        {action_respond_info('Invalid mesh - For SmartABL compatibility')}
    {% else %}
        {action_respond_info('Bilinear Leveling Grid: Dummy result for SmartABL compatibility')}
    {% endif %}

[delayed_gcode smartabl_startup]
initial_duration: 1.
gcode:
    BED_MESH_CLEAR

scmanjarrez, 2023-02-14:
Hi, sorry for the delay, I've been pretty busy.

I've added a [prerelease for 0.1.4](https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4rc0), do you mind checking it? It should have compatibility with Prusa, and being able to ignore gcodes.

iceparr0t, 2023-02-18:
Installed PR 0.14 fine, starting some basic behavior tests to see if ABL will be skipped after the initial mesh.

scmanjarrez, 2023-02-28:
Hi, any problem encountered with version 0.1.4? I've tested it a bit with klipper, but not that much.

scmanjarrez, 2023-03-02:
Thank you!! I'll check that and then prepare the release of 0.1.4

scmanjarrez, 2023-03-02:
I could replicate it only the first time I tested it... I've not being able to replicate it anymore. I think it must be something related to "Save" on settings, for some reason, sometime settings aren't registered and I have to do it multiple times (not related with SmartABL)

scmanjarrez, 2023-03-02:
Closed as released in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4

closed 2023-03-02 by scmanjarrez