scmanjarrez/OctoPrint-SmartABL #5: [Request] Special behavior short short prints

helloITdepartment, 2023-01-04:
### Is your feature request related to a problem? Please describe.

I'm often frustrated when quickly firing up some 15 minute, off the cuff print, and find that because of bed leveling, it's be twice as long until I can have my part. Granted, my 7*7 leveling might be excessive, but still

### Describe the solution you'd like

A section of settings for short prints that override the general settings. There could even be a field for the user to specify what they mean by short, but I think less than 20-30 minutes should be a good default. That way if you plan on firing off a bunch of little prints, you can have those force an updated mesh after every 10 prints, while your normal prints are every 5

scmanjarrez, 2023-01-04:
Mmmm, what do you think about a new button in the side panel to enable/disable "short print mode", triggering ABL only after X prints set in the settings? This would allow switching between modes on the fly without opening, setting and switching between modes in the settings panel

helloITdepartment, 2023-01-04:
I was thinking it could even just autodetect the length of the print and do it based on that, that way you don't need any extra input from the user

scmanjarrez, 2023-01-04:
I'm not understanding this request. What exactly do you want? That prints that are less than 30min iin print time do not trigger ABL? Trigger only if 10 short prints are printed? How would this behave if you print a long-print in between?

scmanjarrez, 2023-04-05:
Closed as no response from user. Reply to reopen

closed 2023-04-05 by scmanjarrez