scmanjarrez/OctoPrint-SmartABL #6: [Request] Ability to add other GCODES to be stripped out if leveling is skipped

1n5aN1aC, 2023-01-04:
### Is your feature request related to a problem? Please describe.

I have a printer which I use G80 to request a mesh bed level, so this plugin currently does not work for me.

Additionally, I use a custom GCODE command which tells the printer to wait for the mesh leveling probe to reach a preset temperature, so it will be more accurate.  (an M command)

### Describe the solution you'd like

I would like the ability to skip both of these commands on this printer if bed leveling is expected to be skipped.

Would it be possible to add a feature where an alternate list of GCODES (and M codes) could be specified, which would be skipped instead if bed leveling is to be skipped?

scmanjarrez, 2023-01-05:
Do you mind sharing and excerpt of gcode showing the lines you want to be stripped?

scmanjarrez, 2023-01-05:
Ok, let me think about how to handle this. Thanks.

scmanjarrez, 2023-01-05:
Are you using marlin? I can't find information about G80 related to bed leveling. How do you load a mesh from eeprom? Using M420?

1n5aN1aC, 2023-01-05:
It's a Prusa MK3S, which unfortunately uses G80 for mesh bed leveling.  My understanding is they used G80 for it before Marlin standardized on G29.

https://reprap.org/wiki/G-code#G80:_Mesh-based_Z_probe

I honestly am not sure about Loading a mesh on these printers, but I've used G81 (also prusa-specific afaik) to print the current mesh status, and it appears to maintain the mesh status from previous prints.

scmanjarrez, 2023-01-05:
Can you check this:
1. Send G80 to generate a new mesh and print the mesh with G81 (write down the values)
2. Shutdown your printer, start it again
3. Print the mesh with G81, and compare the values with the mesh in step 1

Also, I need a dump from octoprint terminal after you "connect" your printer with octoprint (Connect button), I need to check the banner to know when how the output looks like when using Prusa Firmware.

1n5aN1aC, 2023-01-06:
Hmmm, it appears it doesn't actually keep it across reboots. I haven't tested it yet, but I'd wager it might not even save it to EEPROM, and have to be ran at least once upon power on.

I basically never turn it off, so I never noticed that, but outputs below:

Send: G80
Recv: echo:Enqueing to the front: "G28 W0"
{snip}
Send: G81
Recv: Num X,Y: 7,7
Recv: Z search height: 5.00
Recv: Measured points:
Recv:   0.20083  0.18000  0.19917  0.19333  0.16250  0.09583  0.00333
Recv:   0.14500  0.15083  0.14250  0.14083  0.12917  0.08917  0.02917
Recv:   0.13250  0.15083  0.14333  0.13104  0.11500  0.07250  0.00750
Recv:   0.17167  0.13583  0.14250  0.12500  0.06979  0.02583  -0.05167
Recv:   0.20083  0.13083  0.09333  0.05604  0.01333  -0.03000  -0.14583
Recv:   0.22083  0.12333  0.06333  -0.00750  -0.06000  -0.12917  -0.20667
Recv:   0.17333  0.08833  0.02583  -0.01583  -0.10167  -0.20917  -0.30667

{Power Cycle}

Recv: FIRMWARE_NAME:Prusa-Firmware 3.11.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000

Send: G81
Recv: Mesh bed leveling not active.

With that in mind, I would definitely understand it if you wished to not implement this change.  I assume if you did, you would have to implement some sort of additional rules about requiring it upon first print after connecting to a printer, or something like that.

scmanjarrez, 2023-01-06:
If you send M81 after reboot, is there any mesh? Or is empty? (Have you tested G80, M500, reboot and G81?)
> M500 is implemented in PrusaFirmware, so I guess it should work

1n5aN1aC, 2023-01-06:
In the log above, I did run a G81 after reboot, and it simply returns "Mesh Bed Leveling not active"

M81 returns that it's not a valid command.

I just tested, and Running M500 before the reboot makes no difference.

scmanjarrez, 2023-01-16:
Do you mind sharing the output of G80?

dafi87, 2023-01-16:
This is from a Prusa MK3S+

Send: G80
Recv: echo:Enqueing to the front: "G28 W0"
Recv: tmc2130_home_enter(axes_mask=0x04)
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
Recv: tmc2130_home_enter(axes_mask=0x01)
[...]
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv:   0 step=29 mscnt= 477
Recv: tmc2130_goto_step 0 45 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
[...]
Recv:   0 step=34 mscnt= 546
Recv: tmc2130_goto_step 1 1 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
[...]
Recv: LCD status changed
Recv: ok
Send: M113 S9
Recv: ok

scmanjarrez, 2023-01-16:
Does it show the measured points (some messaged are being filtered by the terminal)? In marlin/klipper, you get the measure for each point in the grid something like:
```
Send: BED_MESH_CALIBRATE
Recv: // probe at 52.000,20.000 is z=2.495000
Recv: // probe at 136.000,20.000 is z=2.410000
Recv: // probe at 220.000,20.000 is z=2.335000
Recv: // probe at 220.000,120.000 is z=2.152500
Recv: // probe at 136.000,120.000 is z=2.167500
Recv: // probe at 52.000,120.000 is z=2.247500
Recv: // probe at 52.000,220.000 is z=2.122500
Recv: // probe at 136.000,220.000 is z=2.115000
Recv: // probe at 220.000,220.000 is z=1.997500
Recv: // Mesh Bed Leveling Complete
```

dafi87, 2023-01-16:
Looks like it does not. The filtered messages were only temps and busy processing.

1n5aN1aC, 2023-01-16:
There’s a separate command (G81) to print the current mesh leveling status, which does include the current mesh.

—Joshua Villwock

scmanjarrez, 2023-01-16:
Yes, but I was thinking in a way to know that the current ABL has been completed. I'm using M500 as a flag, but that only works for Marlin.

zandr, 2023-01-30:
It might be necessary to send an M501 to restore a saved mesh. I can test in the morning, I have a print running. :D

That said, I'd be completely happy if this plugin just suppressed G80 if G81 returned a valid mesh (and the other conditions for suppressing ABL are met). I don't often reboot the printer in the middle of a day unless something bad has happened.

scmanjarrez, 2023-01-30:
Ok, I'm reading the documentation from [prusa](https://help.prusa3d.com/article/prusa-specific-g-codes_112173), so I should replace the G28 in prusa with a G28 W in order to skip the bed leveling. Also, if meshes aren't saved, what do you think would be the correct approach? To trigger an ABL everytime octoprint is connected? Then I can take into account the conditions.

zandr, 2023-02-01:
I haven't looked at the code for this plugin, nor do I particularly know how plugins work, so I don't know what the logic and flow looks like, but if possible, I'd do the following:

If you are in a situation where you would ordinarily suppress ABL (the various counters and timers are satisified), then send a G81. If you get back a mesh, then replace G28 with G28 W and skip G80. Otherwise, let them run.

scmanjarrez, 2023-02-14:
Hi, sorry for the delay, I've been pretty busy.

I've added a prerelease for 0.1.4, do you mind checking it? It should have compatibility with Prusa, and being able to ignore gcodes.

scmanjarrez, 2023-02-28:
Any problem encountered with version 0.1.4 and prusa firmware?

1n5aN1aC, 2023-02-28:
Sorry, must have misunderstood the changes you made a couple weeks ago.

Somehow I saw your comment saying Klipper was implemented, and missed that Prusa was as well.

I will test it over the next couple days!

scmanjarrez, 2023-02-28:
Can you upload here the SmartABL.log? And the gcode used if possible (or just the gcode start)

1n5aN1aC, 2023-02-28:
I looked in ~/.octoprint/logs/plugin_SmartABL.log and it's empty (0 bytes)

Am I looking in the correct location?

Also, for my tests, I literally used the gcode I posted above.  No more, just that to make testing really fast.  (I had first printed 2 short (~30 minute) normal prints and it also didn't seem to work, so I manually made the short test gcode)

1n5aN1aC, 2023-02-28:
Ahh, derp!

Will do so and then try 2 consecutive test "prints" and then let you know.

But I have to wait for my current prints to finish first.  <3

scmanjarrez, 2023-02-28:
I'll do a test using you gcode when I get home.

1n5aN1aC, 2023-02-28:
Test prints with debugging:

```
2023-02-28 18:48:53,384 @on_event > Trigger(event=PrintStarted) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 18:48:53,387 @gcode_queuing:ignore > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache={'M500'}, force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:48:53,388 @on_event:print_start >> Mesh query(cmd=M420 V1)
2023-02-28 18:48:53,520 @gcode_queuing:abl > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:48:53,533 @gcode_queuing:abl_trigger >> Sending ['G80', '@SMARTABLSAVE'] > Internal(valid_mesh=False, cache={'M500'}, force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:49:48,530 @at_command:update_state > State(first_time=False, prints=0, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache={'M500'}, force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:49:52,469 @on_event > Trigger(event=PrintDone) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 18:49:52,472 @on_event:print_stop > State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230)

2023-02-28 18:50:14,901 @on_event > Trigger(event=PrintStarted) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 18:50:14,907 @on_event:print_start >> Mesh query(cmd=M420 V1)
2023-02-28 18:50:14,972 @gcode_queuing:ignore > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:50:15,038 @gcode_queuing:abl > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:50:15,054 @gcode_queuing:abl_trigger >> Sending ['G80', '@SMARTABLSAVE'] > Internal(valid_mesh=False, cache={'M500'}, force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:50:58,778 @at_command:update_state > State(first_time=False, prints=0, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache={'M500'}, force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 18:51:02,706 @on_event > Trigger(event=PrintDone) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 18:51:02,708 @on_event:print_stop > State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230)
```

Full gcode of the test file:
```
M860 S33 ; wait until PINDA is >= 33C
G80 ; mesh bed leveling
G28 W ; home all without mesh bed level
```

Both prints I printed skipped the M860.  Both ran full bed leveling, then homed.

scmanjarrez, 2023-02-28:
Some lines in the log file are missing, can you disconnect and connect the printer and post the log lines here? I don't understand why it's using M420 V1. Seems like it's detecting your firmware as Marlin instead of Prusa

1n5aN1aC, 2023-02-28:
I disconnected to serial via octoprint, rebooted the printer, then connected via octoprint.

I then printed the test gcode and this was the result:

```
2023-02-28 19:21:03,344 @process_line:firmware >> marlin
2023-02-28 19:21:09,306 @on_event > Trigger(event=PrintStarted) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 19:21:09,312 @on_event:print_start >> Mesh query(cmd=M420 V1)
2023-02-28 19:21:09,523 @gcode_queuing:ignore > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 19:21:09,603 @gcode_queuing:abl > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 19:21:09,610 @gcode_queuing:abl_trigger >> Sending ['G80', '@SMARTABLSAVE'] > Internal(valid_mesh=False, cache={'M500'}, force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 19:21:57,735 @at_command:update_state > State(first_time=False, prints=0, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache={'M500'}, force_temp=False, firmware=marlin, probed=True, save_allowed=True)
2023-02-28 19:22:01,487 @on_event > Trigger(event=PrintDone) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 19:22:01,489 @on_event:print_stop > State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230)
```

Definitely looks like it's not detecting the firmware correctly.

Here is my serial log on connection:
```
Connecting to port /dev/tty---0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x----------, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Prusa-Firmware 3.11.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:AUTOREPORT_FANS:1
Recv: Cap:AUTOREPORT_POSITION:1
Recv: Cap:EXTENDED_M20:1
Recv: ok
```

scmanjarrez, 2023-02-28:
oh... FIRMWARE_NAME:Prusa-Firmware 3.11.0 based on Marlin. It's detecting Marlin because it's there... let met see what can I do

1n5aN1aC, 2023-02-28:
AHH, that makes sense!

Yeah, Prusa firmware is *mostly* Marlin, so in hindsight that makes sense, lol.

scmanjarrez, 2023-02-28:
Ok, can you test this release? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4rc1. It should detect prusa now

1n5aN1aC, 2023-02-28:
Hey, it detects as Prusa now!

However, it seems to still not skip the leveling:
```
2023-02-28 21:23:55,723 @process_line:detected_firmware >> prusa
2023-02-28 21:24:11,644 @on_event > Trigger(event=PrintStarted) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 21:24:11,652 @on_event:print_start >> Mesh query(cmd=G81)
2023-02-28 21:24:11,710 @gcode_queuing:ignore > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=False, save_allowed=False)
2023-02-28 21:24:11,751 @gcode_queuing:abl > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=False, save_allowed=False)
2023-02-28 21:24:11,760 @gcode_queuing:abl_trigger >> Sending ['G80', '@SMARTABLSAVE'] > Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=False, save_allowed=False)
2023-02-28 21:25:06,829 @at_command:update_state > State(first_time=False, prints=0, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=True, save_allowed=False)
2023-02-28 21:25:06,846 @process_line:valid_mesh > Internal(valid_mesh=True, cache=set(), force_temp=False, firmware=prusa, probed=True, save_allowed=False)
2023-02-28 21:25:10,784 @on_event > Trigger(event=PrintDone) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 21:25:10,786 @on_event:print_stop > State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230)
2023-02-28 21:29:13,906 @on_event > Trigger(event=PrintStarted) || Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False)
2023-02-28 21:29:13,907 @on_event:print_start >> Mesh query(cmd=G81)
2023-02-28 21:29:14,007 @process_line:invalid_mesh > Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=True, save_allowed=False)
2023-02-28 21:29:14,217 @gcode_queuing:ignore > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=True, save_allowed=False)
2023-02-28 21:29:14,244 @gcode_queuing:abl > Settings(cmd_custom=True, custom_gcode=G80, cmd_ignore=True, ignore_gcode=M860, force_days=True, days=1, force_prints=False, prints=5, failed=False, bedtemp=False, hetemp=False) || State(first_time=False, prints=1, last_mesh=28/02/2023, abl_always=False, last_bedtemp=70, last_hetemp=230) || Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=True, save_allowed=False)
2023-02-28 21:29:14,248 @gcode_queuing:abl_trigger >> Sending ['G80', '@SMARTABLSAVE'] > Internal(valid_mesh=False, cache=set(), force_temp=False, firmware=prusa, probed=True, save_allowed=False)
```

scmanjarrez, 2023-02-28:
For some reason it's detecting that the mesh is invalid, can you share the output from G81?

1n5aN1aC, 2023-03-01:
Huh....

I'm wondering if maybe this isn't possible, at least not reliably.

```
Send: N6 G81*22
Recv: Num X,Y: 7,7
Recv: Z search height: 5.00
Recv: Measured points:
Recv:   0.19417  0.17833  0.16667  0.13583  0.10000  0.02083  -0.09333
Recv:   0.14167  0.00917  -0.06833  -0.09917  -0.15167  -0.23583  -0.06250
Recv:   0.13250  0.13667  0.11083  0.03500  0.05167  -0.00417  -0.09083
Recv:   0.16917  0.12167  0.10833  0.07667  0.00979  -0.04833  -0.14250
Recv:   0.19917  0.11833  0.06167  0.01354  -0.04083  -0.10333  -0.22417
Recv:   0.22250  0.13333  0.04583  -0.04333  -0.11083  -0.18917  -0.28167
Recv:   0.17333  0.08667  0.01167  -0.05333  -0.14750  -0.26750  -0.37750
```

But.... later, I ran it again and got:

```
Send: G81
Recv: Mesh bed leveling not active.
Recv: ok
```

I'm....  not sure what caused it, because I then ran the commands one by one manually and it kept the mesh in-memory for several minutes.  When I later tried a print, your plugin worked perfectly, and it skipped leveling, did the print.  But afterwards G81 returned not active.

I think it's SOMEHOW detecting a print is complete and throwing away the mesh?  Doesn't make any sense to me.

scmanjarrez, 2023-03-01:
Mmmm, does prusa throw away bed mesh after every print?

1n5aN1aC, 2023-03-01:
Sorry, been very busy, and will be for a couple days.

Likely will not be able to test more for a few days.

scmanjarrez, 2023-03-02:
Closed as released in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4

closed 2023-03-02 by scmanjarrez