scmanjarrez/OctoPrint-SmartABL #7: [Request] ABL Misfire

hitnrun30, 2023-01-04:
### Is your feature request related to a problem? Please describe.

I always have to redo the ABL if for some reason it misfires, i.e. there is a big cavern in the middle of the bed.

### Describe the solution you'd like

If the variation is off by more than X redo the auto leveling automatically.

scmanjarrez, 2023-01-05:
What do you mean by variaton? Variation to what?

hitnrun30, 2023-01-05:
> What do you mean by variaton? Variation to what?

From Min to Max Value.  Usually if there is a misfire, the value will be something like -2, so if the variation is greater than -1 automatically rerun,

scmanjarrez, 2023-01-05:
You mean from a specific point in the mesh? How would you know the variation without triggering ABL to know the current mesh?

hitnrun30, 2023-01-05:
> You mean from a specific point in the mesh? How would you know the variation without triggering ABL to know the current mesh?

Well I use the bed leveling visualizer, it always shows a Max, Min and a Var(iation)  When you get the values, get the max, min and subtract the max from the min and then have something that checks the threshold, if its breached, rerun the the G29 code

scmanjarrez, 2023-01-05:
I still don't get it. Imagine that I store the mesh, I calculate the variation from min and max as suggested, but then? If the variation is higher than a threshold, then G29 is sent, generaring a new mesh, ok; but min and max in the new mesh will be the same unless you physically change the bed surface.
And if I load the mesh from memory, I won't be able to know the new variation from last print unless I check the new mesh (which imply triggering ABL).

I'll ask jneilliii about this plugin, because I'm not understanding what you want.

hitnrun30, 2023-01-05:
Ok, I get misfires of the ABL all the time, where in the number registers as a hole or visually a purple cone.  if the bed is relatively level the only colors should be green, with maybe some high spots in yellow and low spots in blue.  If there is a purple cone, redo the mesh.  Take the numbers, figure out the max and min, and if the option is set and the threshold is breached, automatically run a G29.

scmanjarrez, 2023-01-05:
Ok, I've talked with jneilliii, so you want to trigger ABL once more if the first triggered ABL reports a difference between min and max point superior to a threshold, is that correct? What should happen if the second ABL also reports the same difference? I guess you only want the second trigger only to assure that the difference wasn't a fluke, right?

hitnrun30, 2023-01-06:
Technically, or the feature could either have it run x times and have a drop down for the amount of times, and 0 would be until it is working correctly (infinite).

hitnrun30, 2023-01-09:
Just so you don't think I'm lying.
[image]

beffjessen, 2023-02-13:
I would be interested in a related feature: I've noticed that my bed will drift out of parallel over time, and at some point the bed will get so far out that I need to manually tram it back to within some tolerance (i.e., the same variation calculation referenced above). It would be awesome if this plugin enabled the user to:
- Run ABL before every print
- Check that the variation in the bed mesh is within a user-specified tolerance
- If the variation exceeds the tolerance threshold, do something - prompt the user before continuing the print; abort the print; even just do nothing, but raise a flag so the user could tram before the next print