scmanjarrez/OctoPrint-SmartABL #8: Question regarding functions

Visthum, 2023-01-08:
Hello, I've gotten the plugin to work just once and I have it set to abl after 6 successfully prints. But now, I can't get it to abl after 6 prints. My question is, I'm using prusaslicer and I do not have G29 in the gcode as I rarely have to abl my bed, I have G28 followed by M420 s1 z10.
Is it required to include G29 into custom g-code for this to work?

scmanjarrez, 2023-01-08:
Mmmm, yes. Enable the checkbox in settings, so the plugin doesn't send m420 again and instead send G29

Visthum, 2023-01-08:
> Mmmm, yes. Enable the checkbox in settings, so the plugin doesn't send m420 again and instead send G29 

[image]

This you mean?

Do I also have to add G29 in prusaslicer?

scmanjarrez, 2023-01-08:
Correct, that should work. You don't need to modify your slicer settings. SmartABL should recognize M420 and trigger ABL according to your settings.

Visthum, 2023-01-08:
> Correct, that should work. You don't need to modify your slicer settings. SmartABL should recognize M420 and trigger ABL according to your settings.

Thank you! I will test it with a couple of fast prints and see if it works :) it's a great plugin

Visthum, 2023-01-08:
Okay yes it works! Thank you :) however I'm wondering about one other thing.
When the abl has finished, it goes straight to the printing process and therefore i can't manually adjust the bed or set the offset like you can when you do abl normally.
Im sure there's a command for this.

scmanjarrez, 2023-01-08:
Mmmm, that's not what I do. I usually send G29 to generate the mesh and start printing, the printer compensates the Z-height according to the mesh generated with G29, without physically modifying the bed. I only do manual leveling using tramming wizard (marlin), but I do it occasionally, when the points in the mesh are higher/lower than 0.1mm. 

How do you use ABL?

Visthum, 2023-01-08:
I see. Well it's not a problem at all but this is how I usually do it :) 

I usually run the abl and then I check each corners with a 0.1 blade and adjust if needed. 
I know its not needed each time and I've actually started to not do this each time and trust the probe but sometimes I do this, because I have the probe at a bad location (ender s1 pro) it won't probe the whole bed and thus miss some of the bad spots. 

Every now and then when needed, before I manually run the abl, I first use a digital dial indicator and make sure the bed is 100% level and then I run the abl and then I do manual checking after that. Since I started doing this, my first layers are amazing. I have the silicone rubber spacer instead of the springs so I do this maybe only once a month.

**And btw, would it be possible to include some kind of counter that tells you how many prints before it will trigger the abl? **

scmanjarrez, 2023-01-08:
Noted! I'll add that to the next release

scmanjarrez, 2023-02-14:
Hi, sorry for the delay, I've been pretty busy.

I've added a [prerelease for 0.1.4](https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4rc0), do you mind checking it? It shows a counter under the print information with the number of prints until ABL is triggered.

Visthum, 2023-02-15:
> Hi, sorry for the delay, I've been pretty busy.
> 
> I've added a [prerelease for 0.1.4](https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4rc0), do you mind checking it? It shows a counter under the print information with the number of prints until ABL is triggered. I've fixed the bug with M420 not triggering ABL.

I'll check it out! And write a follow up if I encounter anything weird. Out of all the plugins I have, I think yours is the most useful so keep up the fantastic work and take care :)

scmanjarrez, 2023-03-02:
Closed as released in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4

closed 2023-03-02 by scmanjarrez