scmanjarrez/OctoPrint-SmartABL #9: ABL every time you start printer?

AndyShap2020, 2023-01-19:
### Is your feature request related to a problem? Please describe.

I have my gcode set to default. I am forcing an ABL after 10days. I am forcing an ABL after 5 prints. I am set to ignore failed prints.

All seems to work fine, but I turn my printer off if I am not immediately doing another print.  Is it normal that is does an ABL at the first print, EVERY time  I power on? If I power off and on 4 times in the same day, it will do an ABL at the start of each of those 4 prints. 

### Describe the solution you'd like

I'd like it to ignore power on/off and follow the rules set in the options.

scmanjarrez, 2023-01-19:
Do you mind enabling logs file and sharing octoprint_SmartABL.logs? Also, are you using marlin? Do you have eeprom enabled?

AndyShap2020, 2023-01-20:
[octoprint-logs.zip](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10463603/octoprint-logs.zip)
Yes, I am using Marlin. I am using MRiscoc's firmware. I am pretty sure eeprom is enabled. How do I know for sure?

Is the behavior I am describing normal?

AndyShap2020, 2023-01-20:
I am just looking at the log. It is empty. Hang tight, I will verify logging is enabled and post back.

AndyShap2020, 2023-01-20:
Sorry about that. I have enabled logging and will let it run for a day or two and get back to you.

scmanjarrez, 2023-01-20:
You can verify that eeprom is enabled by doing a G29 (note the values) and then M500 manually, then query the eeprom information sendi M503, that command should report the same mesh points returned by G29

AndyShap2020, 2023-01-31:
Thank you for your patience (I've had some big print jobs running).  Here are the results of my eeprom commands as well as the Sm
[eeprom results.txt](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10550205/eeprom.results.txt)
[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10550206/plugin_SmartABL.log)
artABL logs

scmanjarrez, 2023-01-31:
Do you mind sharing your start gcode?

AndyShap2020, 2023-01-31:
Wow! Thank you for the quick response!  G-Code attached
[Start G-Code.txt](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10550926/Start.G-Code.txt)

scmanjarrez, 2023-01-31:
Can you share the output from M420 V1? 

Also, enable `custom` under Leveling command in SmartABL settings. Currently, the plugin is sending "M420 S1 Z2, M500", which is recovering the mesh from eeprom, not triggering ABL. I don't find any G29 in the logs. 

In the logs, it seems that you have enabled failed prints in the counter.

AndyShap2020, 2023-02-03:
Thank you again for your patience.  I'm running some 24 hour jobs, so I need to wait for the jobs to finish.

Attached are results from G29 command and M420 V1.
[G29.txt](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10582132/G29.txt)
[M420 V1.txt](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10582133/M420.V1.txt)

I was thinking CUSTOM meant "don't do the default action, do this instead". And I thought the default WAS G29.  I have enabled custom now.  Also, what I want is for failed prints to NOT be taken into account in my "every 5 prints do an ABL" request. Would I UNCHECK the failed prints option to achieve this?

scmanjarrez, 2023-02-03:
Yes, checked means take into account failed prints. The default action is "use my gcode instead (because someone was using g29 with a lot of arguments, however that doesn't take into account M420 which means "load mesh instead ")

AndyShap2020, 2023-02-03:
OK.  I have the M420 command in there because I DO NOT want to run an ABL every time I print.  So, the m420 says use the stored mesh level info (I'm talking out loud here, not trying to teach you something!).  Are you saying I should remove the M420 command from my start Gcode and let SmartABL handle it? And if so, where is it getting the mesh level info in between fresh ABL's triggered by SmartABL?

scmanjarrez, 2023-02-03:
Nono, it's just a bug in my code. I didn't even think about that case. But you can use custom gcode while I add that fix and some other functions pending. Without the custom gcode, the plugin monitor G29* and M420, if it detects any of these, it sents it back to the printer if ABL should trigger (which is incorrect in the case of M420, it should sent at least G29, sending M420 back does nothing), otherwise M420 is sent instead.  I'm not in home now. I'll check the logs later. But when  I checked them, (I think) the ABL wasn't from me, because the plugin was sending M420, not G29

AndyShap2020, 2023-02-03:
Sorry, a little confusing...  I think you are telling me to just leave it all as is for now?

P.S.  I'm happy to do some beta testing for you if you would like. You certainly have been more then helpful and responsive!

scmanjarrez, 2023-02-03:
That would be helpful, I've switched to Klipper few weeks ago and I need to go back and forth to test marlin. For now, uncheck failedPrints and check custom (with G29 in the box). The plugin will manage your bed leveling according to the conditions.

AndyShap2020, 2023-02-09:
FYI, I've followed directions per your most recent comment. It still does an ABL Every time I turn on the printer.

scmanjarrez, 2023-02-09:
I needs the logs. Disabling the plugin doesn't trigger ABL?

scmanjarrez, 2023-02-14:
Hi, sorry for the delay, I've been pretty busy.

I've added a [prerelease for 0.1.4](https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.1.4rc0), do you mind checking it? I've fixed the bug with M420 not triggering ABL when required.

scmanjarrez, 2023-02-14:
Configure it as you want, you don't need to check FAILED in settings. Leave M420 in gcode, yes, the plugin should use g29 instead.

scmanjarrez, 2023-02-20:
Did you manage to install it? Is it working now? 

If you still don't know how to install, go to settings, plugin manager, get more, and paste this link under the box "... from url" https://github.com/scmanjarrez/OctoPrint-SmartABL/archive/refs/tags/0.1.4rc0.zip and click install

AndyShap2020, 2023-02-20:
Hi Sergio,  Yes, I got it installed.  I am not sure how it is working (thus I haven't responded yet).  I have it set for 10 days or 5 prints.  I had a 21 hour print and maybe 1 or two others. It did an ABL on the next print. So, not 10 days (1 day) and not 5 prints (about 3 prints).  I really want to do a few more prints before I say anything.

P.S.  Does the failed prints option include stopped prints? I pretty much never have a failed print where the printer stops.  I have MANY prints that aren't going well, so I stop the print.  Hopefully, stopped prints are included in failed prints.

scmanjarrez, 2023-02-20:
Yes, cancelled and stopped prints. Only successful prints count if disabled.

AndyShap2020, 2023-02-20:
P.S. As noted on the screen print you just loaded, my Octoprint DOES NOT show a Green ABL restricted. It is a blank square box. It DOES show ABL Always.  I can't tell if ABL always is selected though. Does it change color or anything when selected? 

[image]

scmanjarrez, 2023-02-20:
lol, yes, it was supposed to be green, and blue when abl always is pressed.
Like this in v0.1.4
[image]
[image]

Do you have any theme related plugin?

AndyShap2020, 2023-02-20:
I have "Themeify" plugin enabled.  I'd hate to lose it.  I have disabled it and the SmartABL portion looks correct.  I assume the middle portion tells me how many prints since last ABL?  Is there any way to incorporate SmartABL to be used alongside Themeify?

I'll keep an eye out to see if the issues I have been having are resolved by disabling Themeify.

scmanjarrez, 2023-02-20:
I have enabled themeify as well, and it works for me. Do you have custom css? What is your themeify configuration? I don't think themeify is the one triggering ABL

AndyShap2020, 2023-02-20:
[image]
[image]

scmanjarrez, 2023-02-28:
Sorry. I found the problem, it seems like the culprit is the Material Light UI theme. Using the others kinda work. I think you can "disable" the theme but still being able to use custom css (that's what I do, because Material UI lose lot of button colors).

AndyShap2020, 2023-02-28:
I'll take a look.  I was doing testing of the new release, I got to 4/5 prints and then, 20 hours into a 22 hour print, our power shut off.   Octoprint reset the print count back to 0. :-( 

I'll keep testing.

AndyShap2020, 2023-03-04:
I don't know if we are there yet. I started a print and it did an ABL. The print failed, and I started a new print. It did an ABL again. That second print failed and I started a third ABL again. I don't think it should have been doing an ABL each time (The count DID remain at 0/5, as it should since I never got a successful print.

Also, Can I get a screen print of your CSS entries, I'd like to get themify back if I can. 

Thanks!

scmanjarrez, 2023-03-04:
It's really difficult to debug a problem without a log file... I don't use custom CSS, just the default one (and discoranged theme)

AndyShap2020, 2023-03-04:
[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10889831/plugin_SmartABL.log)

scmanjarrez, 2023-03-04:
Ok, I think I have detected the problem. I'll release a hotfix. Also, take into account that you have PrintFailed checked, that means that every stopped/cancelled print is increasing the counter.

AndyShap2020, 2023-03-04:
A little confusing... I DO have PrintFailed checked. I thought that means DO NOT use failed prints in the count. But you are saying that when checked it means DO count failed prints in the count?

AndyShap2020, 2023-03-04:
P.S. I can't believe how responsive you are! I am still more than happy to continue beta testing with you.

scmanjarrez, 2023-03-05:
I've released a new version, it should pop up in the following hours in your octoprint instance.

scmanjarrez, 2023-03-13:
hi, have you encountered the same error after last update (ABL still triggering)? Also, I'll try to add custom CSS rules at least for my buttons so it works even with themeify material design.

AndyShap2020, 2023-03-13:
It’s been a bit of a crazy week. I have not yet had a chance to install the new update. I will do it today and run several small jobs and get back to you.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-03-13:
Same, don't worry, when you have time. Thanks!

AndyShap2020, 2023-03-14:
I’m just getting started on testing.  Octoprint is coming up saying 0.5.0 is ready to install. Is this the latest release I should be testing?

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-03-14:
Yes, that's the latest version

scmanjarrez, 2023-03-14:
Yes, I'm planning to create a custom "theme" for these buttons in order to fix that, but I've been very busy lately, sorry.

AndyShap2020, 2023-03-14:
Don’t apologize, I’m just making sure I understood what you said previously.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-03-14:
Should open this request in a new thread?: As of now, SmartABL only heats the bed, then when up to temp, it does an ABL.  I’d like it to heat the head at the same time, so I don’t need to wait after an ABL.  Can I just add this in the custom GCODE?

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-03-14:
Yes, you can add several gcodes, just separate them with a comma. Btw, SmartABL has nothing to do with hotend/bed, it only cares about bed leveling commands. If the bed/hotend isn't heating, it should be in your start gcode.

AndyShap2020, 2023-03-14:
No, it does heat up the bed before abl, but I'd like it to heat up the hotend as well. I'll add gcode in smartabl.

First print started, did an abl as expected.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.******@***.***>
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-03-14:
Yes, but it's a misconfiguration in your slicer. You should use M140 and M104 to start preheating, then M190 and M109 to wait for the temperature reaches the required value.

AndyShap2020, 2023-03-16:
Unfortunately, we aren't quite there yet.  I started a print on Monday and it did an ABL and started a print.  This is now my second print (2 days later, I have it set for 5 prints or 10days).  And it is starting an ABL.  Logs attached.
[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10995173/plugin_SmartABL.log)

scmanjarrez, 2023-03-16:
Ok. I got a clue from the logs. For some reason, the event PrintStarted from octoprint is delayed. The plugin is receiving the event PrintStarted (the one I use to start the mesh check) after receiving G29, so the check returns "incorrect bed mesh". The normal behaviour is, the plugin receives PrintStarted, then I query the bed mesh with M420 V1, if it returns the mesh, I set a flag. Then when I receive G29 I check that flag, if all went good (the flag exists), ABL is skipped. If the flag is missing, ABL is triggered (your case)

scmanjarrez, 2023-03-18:
Hi, I've pre-released a new version with a (I hope) fix for this error. Can you check it?

AndyShap2020, 2023-03-18:
I should be able to do it today.  Print 2/5 also did an abl.  Where is the software? Is it already on GitHub?

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.******@***.***>
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-03-18:
Yes, I haven't released it yet to octoprint. But you can find the pre-release here https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.1rc1. Also, check this plugin https://github.com/LazeMSS/OctoPrint-UICustomizer, far better than themeify (btw, I've added the fix to your themeify problem)

AndyShap2020, 2023-03-18:
Installed update and starting first print now.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-03-18:
I’ve enabled Themify, I can now read RESTRICTED, but it is Grey and not green as when Themify is diabled.  So, Grey versus white is still hard to tell if restricted is on or off.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-03-19:
Just starting my 3rd print. No ABL, this is the first time this is working!

[image]
[image]
[image]

scmanjarrez, 2023-03-19:
Yes! I'm working on the css fix.

scmanjarrez, 2023-03-19:
Can you check this version? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.1rc2 It should give some colors to your buttons :)
Also, check [UICustomizer ](https://plugins.octoprint.org/plugins/uicustomizer/)plugin, is the next-generation themeify, way better and maintained.

AndyShap2020, 2023-03-19:
I am just heading out of town today. I’ll be back tomorrow and have prints to start as soon as I get back. I’ll let you know tomorrow afternoon.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-03-21:
Green color is back! Started my fourth print- no ABL!

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-03-21:
Can you share a screenshot from SmartABL settings? Maybe you have custom gcode enabled? Also, can you share your start gcode?

[11 more replies]

closed 2023-03-25 by scmanjarrez