scmanjarrez/OctoPrint-SmartABL #10: ABL every time (N3P)

boehser-enkel, 2023-02-20:
### The problem

ABL is done every time regardless of plugin settings

### Version of SmartABL

0.1.3

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

Docker (Raspbian)

### Printer model & used firmware incl. version

Elegoo Neptune 3 Pro (1.5.3)

### Browser and version of browser, operating system running browser

Any

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230220113238.zip](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10783934/octoprint-systeminfo-20230220113238.zip)
[octoprint-logs.zip](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/10783935/octoprint-logs.zip)

scmanjarrez, 2023-02-20:
SmartABL.log is empty. You need to enable DEBUG mode in logging (settings). Also, which firmware are you using? Elegoo Neptune 3 Pro (1.5.3) isn't a firmware.

boehser-enkel, 2023-02-22:
Sorry missed a number: 1.1.5.3
I enabled logging and will test again

boehser-enkel, 2023-02-22:
It's the Neptune 3 Pro. Yes Marlin.
I use G29. I think it's enabled. At least the mesh values are saved(?)
`Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.1.1 (Feb  2 2023 16:05:30) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:1
Recv: Cap:CASE_LIGHT_BRIGHTNESS:1
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:HOST_ACTION_COMMANDS:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:LFN_WRITE:0
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0
Recv: Cap:EXTENDED_M20:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: ok
Recv:  T:22.81 /0.00 B:23.44 /0.00 @:0 B@:0
Send: M21
Recv: echo:SD card ok
Recv: ok
Send: M155 S2
Recv: ok
Send: M876 P1
Recv: echo:M113 S2
Recv: ok`

scmanjarrez, 2023-02-28:
Hi, did you enabled the logs? The file you uploaded was empty.

scmanjarrez, 2023-04-05:
Closed as no response from user. Reply to reopen

closed 2023-04-05 by scmanjarrez