scmanjarrez/OctoPrint-SmartABL #11: Z Steppers Auto-Alignment (G34)

iiKelZenii, 2023-03-06:
### Describe the solution you'd like

Please can your add  "Z Steppers Auto-Alignment" feature support?
I have dual Z axis motor, and after power off Z steppers, or power off printer, Z axis can move down. I wold like to use "Z Steppers Auto-Alignment" marlin feature to fix this problem before bed leveling.
Or may be can your add custom G-code edit menu for bed leveling? Not a single command. 
Thanks.

scmanjarrez, 2023-03-06:
How does it work? You send G34 and then G29?

iiKelZenii, 2023-03-07:
> How does it work? You send G34 and then G29?

I send G28 then G34 and then G29. 

> I can make the custom gcode to receive a list of gcodes instead, is that ok?

Yes, I think it would be great.

scmanjarrez, 2023-03-07:
Ok, I'll add that asap, thanks!

scmanjarrez, 2023-03-07:
Released on 0.5.0

closed 2023-03-07 by scmanjarrez

iiKelZenii, 2023-03-08:
I test it. Works fine for me. Thanks.