scmanjarrez/OctoPrint-SmartABL #12: SmartABL disabling due to "unsupported firmware" but running Klipper

iceparr0t, 2023-03-22:
### The problem

When I start 

Upon upgrading from the 0.14 pre-release for Klipper testing to the new 0.5.0 release, I now get a message that my firmware is unsupported and SmartABL plugin is disabled.

Using Klipper with OctoKlipper plugin.

Message appears when connecting OctoPi to Klipper.

[image]

Terminal output at connection to Klipper is:

```
Performing autodetection with 7 port/baudrate candidates: /tmp/printer@115200, /tmp/printer@250000, /tmp/printer@230400, /tmp/printer@57600, /tmp/printer@38400, /tmp/printer@19200, /tmp/printer@9600
Trying port /tmp/printer, baudrate 115200
Connecting to port /tmp/printer, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xa9edf5b0, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Detecting serial connection" to "Operational"
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.9.1-617-gf2421005 FIRMWARE_NAME:Klipper
```

### Version of SmartABL

0.5.0

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3 running Klipper 0.

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

scmanjarrez, 2023-03-22:
Oh, I can see the error. I'll fix it asap. Thank you!

scmanjarrez, 2023-03-22:
Can you check this version, please? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.1rc4

iceparr0t, 2023-03-22:
Error is gone with 0.5.1rc4 and SmartABL plugin stays enabled.   First print initiated a full measurement as expected.   Will do some basic regression testing over my next couple prints.

iceparr0t, 2023-03-22:
Octoprint now waits infinitely after the BED_MESH_OUTPUT, when it should be proceeding with the rest of the gcode.

`Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 1.122500 0.655000 0.380000 0.152500 -0.037500
Recv: // 0.827500 0.522500 0.322500 0.177500 -0.070000
Recv: // 0.487500 0.182500 0.072500 -0.025000 -0.090000
Recv: // 0.145000 0.000000 -0.017500 0.007500 -0.107500
Recv: // -0.137500 -0.320000 -0.260000 -0.175000 -0.102500
Recv: Mesh X,Y: 13,13
Recv: Search Height: 5
Recv: Mesh Offsets: X=0.0000, Y=0.0000
Recv: Mesh Average: 0.14
Recv: Mesh Range: min=-0.3200 max=1.1225
Recv: Interpolation Algorithm: bicubic
Recv: Measured points:
Recv:   -0.137500  -0.188407  -0.271759  -0.320000  -0.310222  -0.281667  -0.260000  -0.236000  -0.199556  -0.175000  -0.152611  -0.121111  -0.102500
Recv:   -0.065148  -0.112295  -0.189637  -0.235000  -0.230077  -0.210470  -0.194870  -0.174696  -0.143796  -0.124500  -0.116832  -0.108670  -0.104130
Recv:   0.057426  0.016072  -0.052045  -0.093111  -0.096463  -0.091972  -0.086630  -0.073120  -0.051884  -0.041556  -0.058060  -0.088140  -0.106648
Recv:   0.145000  0.105519  0.040259  0.000000  -0.009463  -0.015593  -0.017500  -0.009463  0.003796  0.007500  -0.023278  -0.075611  -0.107500
Recv:   0.242204  0.191398  0.107216  0.054463  0.037035  0.020161  0.010648  0.008249  0.006081  0.001000  -0.027830  -0.074747  -0.103148
Recv:   0.387741  0.317551  0.201033  0.127148  0.097583  0.064820  0.044019  0.025304  -0.000988  -0.019389  -0.042118  -0.075552  -0.095463
Recv:   0.487500  0.405537  0.269352  0.182500  0.144759  0.101019  0.072500  0.043481  0.002019  -0.025000  -0.045704  -0.073630  -0.090000
Recv:   0.586463  0.504250  0.367400  0.279130  0.234341  0.178981  0.142833  0.109540  0.062228  0.029907  -0.004790  -0.054803  -0.084481
Recv:   0.730648  0.650310  0.516185  0.428093  0.373253  0.300842  0.253611  0.215381  0.161554  0.122259  0.063979  -0.023743  -0.076185
Recv:   0.827500  0.746870  0.612019  0.522500  0.460796  0.377093  0.322500  0.280500  0.221611  0.177500  0.105370  -0.004315  -0.070000
Recv:   0.918426  0.825934  0.671302  0.568889  0.499857  0.406676  0.345667  0.296242  0.226691  0.176648  0.106644  0.001992  -0.060500
Recv:   1.046685  0.934551  0.747224  0.623722  0.544134  0.437872  0.367944  0.307107  0.221087  0.162352  0.099126  0.007977  -0.046111
Recv:   1.122500  0.998444  0.791278  0.655000  0.569148  0.455185  0.380000  0.312315  0.216407  0.152500  0.093685  0.011204  -0.037500
Recv: 
Recv: ok
[...]`

scmanjarrez, 2023-03-22:
Can you upload plugin_SmartABL.log file?

iceparr0t, 2023-03-22:
[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11044680/plugin_SmartABL.log)

scmanjarrez, 2023-03-22:
I've already found the problem, can you check this version? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.1rc5

iceparr0t, 2023-03-22:
rc5 changes appear to have resolved the issue created in rc4 fix.   Recommend closing as the original bug is now resolved.

scmanjarrez, 2023-03-22:
I'll keep it open until 0.5.1 is released. Thanks for opening the issue!

scmanjarrez, 2023-03-25:
Fixed on https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.1

closed 2023-03-25 by scmanjarrez