scmanjarrez/OctoPrint-SmartABL #13: M420 S1 and then print just stalls

tkunchick, 2023-04-01:
### The problem

I don't know if I have something configured wrong or not. but since last update, when trying to print with unrestricted able with unified bed leveling, watching terminal, when I start a print, I see it go through the motions, once it hits abl, in this case, g29 p1, instead it runs M420 S1, and then just stops

### Version of SmartABL

SmartABL (0.5.1)

### Version of OctoPrint

 OctoPrint 1.8.7 

### Operating system running OctoPrint

OctoPrint 1.8.7 Python 3.9.2 OctoPi 1.0.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

tkunchick, 2023-04-01:
disabling plugin allows print to process normally.

scmanjarrez, 2023-04-01:
Can you upload your plugin_SmartABL.log? Also, this behavior only occurs when using ABL buttons?

scmanjarrez, 2023-04-01:
Also, a screenshot of your settings if possible.

tkunchick, 2023-04-01:
The log file is a 0 byte file currently. should I set debug level on the log? Sorry if I don't understand, never been much on the programming side of IT. Current print has 3-4 hours so any changes will take at least that long.
[image]

As far as settings, I normally run both printers on 5 prints or bed temp change. though I recently changed that one jus on the main screen to always run ABL due to the bed being warped/skewed and any time I move I remove the glass it changes the level.  
[image]

scmanjarrez, 2023-04-01:
Yes, you need to enable debugging under settings -> logging, then set SmartABL to DEBUG. Can you share your starting gcode? I can do my own tests using your starting gcode using the config presented in the screenshot

tkunchick, 2023-04-01:
Had it on my desktop after watching the m420 go by in temerinal wondering where the heck it came from :-)

`; Postprocessed by [ArcWelder](https://github.com/FormerLurker/ArcWelderLib)
; Copyright(C) 2020 - Brad Hochgesang
; arc_welder_resolution_mm = 0.05
; arc_welder_g90_influences_extruder = True
;FLAVOR:Marlin
;TIME:24014
;Filament used: 22.2745m
;Layer height: 0.2
;MINX:79.933
;MINY:59.679
;MINZ:0.2
;MAXX:150.068
;MAXY:165.32
;MAXZ:40
;POSTPROCESSED
;Generated with Cura_SteamEngine 5.3.0
; thumbnail begin 230x180 4252
; /9j/4AAQSkZJRgABAQEASABIAAD/2wBDAAgGBgcGBQgHBwcJCQgKDBQNDAsLDBkSEw8UHRofHh0aHB
; wgJC4nICIsIxwcKDcpLDAxNDQ0Hyc5PTgyPC4zNDL/2wBDAQkJCQwLDBgNDRgyIRwhMjIyMjIyMjIy
; MjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjL/wAARCAC0AOYDASIAAhEBAx
; EB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQAAAF9AQID
; AAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3ODk6Q0RFRk
; dISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWmp6ipqrKztLW2
; t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEAAwEBAQEBAQEBAQ
; AAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSExBhJBUQdhcRMiMoEI
; FEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElKU1RVVldYWVpjZGVmZ2
; hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3uLm6wsPExcbHyMnK0tPU
; 1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwD5/ooooAKKKKACiiigAooooAKKKK
; ACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKK
; KKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooA
; KKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAoqeK0uJ4mkiiZ1U4OKh
; ZSpwwIPoaB2YlFFFAgooooAKKKKACiiigAooooAKK0tL0O91Z/3Ee2LPMr8KP8fwrtdO8JadZxjzox
; cy92kHH4ColNROuhgqtbVKy7s86RGkYKilmPYDJrRt/D2rXWPLsZQD3cbB+tenw28Nuu2GGONfRFA/
; lUuKzdbsj0IZTH7cvuOAg8DX8gzNPBF7DLGq114P1a3yUjSdR3jbn8jivScUuKn2sjd5XQasrnjc9r
; cWr7LiCSJvR1K/zqKvZ3iSVCkiK6nqGGRWVdeFdHu8k2oiY/xQnb+nT9KtVl1OWplE18Er+p5bRXbX
; fgDqbO9+izL/AFH+FYV34V1i0yTaNKo/ihO/9Bz+lWpxfU4amCr094/qY1FOeN4nKSIyMOoYYIptWc
; oUUUUAFFFFABRRRQB2Nhbi1s44xjOMsR3JqWa1huBiWNW+orl7TVrm0wu7zIx/C39DW7aavbXWFLeX
; J/df+hrNpo76dSElylW58Po2Wgcr7HmsmfTrm3J3RlgO6812IoZVcYYA0KTCWHi9tDg6K6+50qC4BJ
; QE/r+dZM+glSdku30Djg/iKpSRzSoTiY1FTTWs9uf3kZA9RyPzqGqMWrBRTkR5HCIrMzHAVRkmup0n
; wZNNtl1FjEnXyl+8fqe1JyS3NqNCpWdoI5yzsrm/mENrC0j+g7fU9q7PSfBkFvtl1BhPJ1EY+4Pr61
; 0lpZ29lCIbaFYox2Udfr61PWEqrex7WHy6FPWer/AaqrFHhVCooxgDAFOBDAEHIPQ1Isrqm0Hj6U0C
; sj0rLoJilxS4oZljXc5wPU0DFxRilxS4pFWDFLiilxQOwmKXFLRigdiGe0t7pNlxBHKvo6hv51iXfg
; rSLnJjjkt2PeJuPyOa6PFGKFJrYipQp1Pjimef3fw/ukybS7ilH92QFT/WsG88PatY5M9jLtH8SDeP
; zGa9fxS1oq0lucNTKKEvhujwzpRXtN3pVhfA/arOGUn+JkGfz61xfirwpa2NoLrTkdSuS8ZYsNvqM8
; 1rGqm7Hm4jKqlKLmndL7ziqKKK1PLCiiigC7aardWmAr70/uPyP/rVvWmtWtzhXPlSejHj865Sik4p
; msK0oHfA0pAIwRkVxlpqd1Z4Eb7k/uNyP/rVvWeu21xhZf3L/wC0eD+NQ4tHZDEQlo9C9JaRuOPl/l
; Wbc6NE/Jj2/wC0lbIIIyDkU6ldo0lSjIXRBpenRALEEkI+aZjuJ/qPyxXRI6SKGRgynoQciqGg6dpm
; p3bWV8jI8o/dSo2CG9PQ5+lXrzwJq2nMZtKuRcL1252N+R4NZySbO2jOpCCtG68v8h+KdishNSvLSb
; 7PqVo8cg65G1vyPX8K0be7t7ofuZVY916EfhUOLR1060J7PUnxS4oxS1JvYMU4MxXaTn60mKWgpBS4
; oxQRkYBI9xQMY0saOqs2CxwOKkpcZAzzj1paQJMTFLS4pcUFWExS0UuKB2DFLijFLikMMVmanguikZ
; G3kGtJmC9az78IVM8kqxog5LnAA+tVHcyr/A0eX69pZ02+OwfuJPmjPp6j8KK1tZ8Q2M5EEdst0iNn
; c+QufbvRXXFu2qPk60KXO+WWhylFFFWcoUUUUAFFFFAFu01G5syPKkO3+43Ires9ft58LOPJf1P3T+
; NctRScUzWFacNj0KKUo6SxPhlIZWU9D2Ir1zRNTTVdMiuVwHI2yKOzDr/j+NfNVpqFzZNmGQhe6nkH
; 8K7Xwx8QRpDyCWLCyAZByVyO/HIrKUGephMdBO0tLnttxZ2t9CYru3injP8ADIoIrltS+G9hcEy6bc
; SWco5Ckl0/xH5mqdv8R0uF3Q2cUg/2Z8/0q2PiEo/5hp/7/f8A2NQro7pujU31OcvNM8SeHwWu7b7X
; ar1ljO4AfXqPxFNtdZs7nAL+U/8Adk4/XpWxP4jOqN++n2jtGeAKqXGn2l1lpIVLH+IcE/41La6o1p
; 0qkVeE7rs/8yQU7FZj6dd2p3WFwoXvHIOD/h+GKVdUeAhdQtng/wCmijch/EVNuxr7Xl0qK35feaVL
; imxSxzIHidXU91OafUm611QUuKXFFBQUxw5bAXKHuDUmKWgLXIIbVYZGdS2WGCM8VYxRTGlA4Xn3o3
; BJRQ8kAZJqJpf7vA9aydW1+x0pc3Mu6XHyxJyx/Dt+NcDrHiq+1TdGjfZ7Y/8ALNDyR7nvVwpuRw4r
; MKdDTd9jr9Y8X2WnbooCLm5HGFPyqfc/4VwWp6zfatJuupiVBysa8Kv0FUKK6YwUT57EY2rX0bsuwU
; UUVZyBRRRQAUUUUAFFFFABRRRQAUUUUAOR3jYMjMrDoVODWnbeIb63wHcTL6OOfzrKopNJlRnKPws6
; 228S2kuBMrQt7/MPzFbdpqBK77W4DL/stkV5vTo5HicNG7Iw7qcGpcEzrp42cdz1iHWGHE0YPutXor
; 23nGA4yf4W4ry228Q3sGBIVmX/AGhz+YrYtvEdnNgShoW9xkfmKydI9Klmaejf3nbPptuX8yINBJ/f
; iO39OhpytewcOEuU9R8r/l0P6VhWuovsDW84dPY7hWgmtIo/0hQo/vA/41DizshXpPVe7/X3GrFcxS
; sFyVf+44wfy71PWDP4i0dVxJdI3ttJ/pVJvGul25xG00g9MAj9TmlySfQ0+u0Y/FNHWU1nC+5rip/i
; FD0htGPuX/8ArVmz+PLxlIgt4UPqwJx+tNUpMynmuGjs7/I7m+1G3soDNdzpFGPU9foO9cPrHjae43
; Q6aphj6ea33z9PSuZvL65v5zNdTPK57sen0Haq9bRpJbnkYnM6lXSGi/Ec7vI7PIzO7HJZjkk02iit
; TzAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAUMy9CR9DQXY9WJ/GkooAKK
[... 37 lines cut ...]
G29 P3    ; infer the rest of the mesh values again
;G29 L0 ; load a valid mesh from slot 0
;G29 A  ; active the UBL system
;M420 S; to turn leveling on
;cancel screens waiting for user simple confirmation
C108
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X7.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X7.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X7.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X7.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X12 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G92 E0
G92 E0
G1 F3000 E-0.5
;LAYER_COUNT:202
;LAYER:0
M117 DASHBOARD_LAYER_INDICATOR 1
M117 DASHBOARD_LAYER_INDICATOR 1
M117 DASHBOARD_LAYER_INDICATOR 1
M117 Printing Layer 0 of 201 ikea leg connector
M107
G0 F6000 X87.283 Y73.271 Z0.2`

tkunchick, 2023-04-01:
Watching the terminal, it would hit the m420 s, report a valid mesh, then just sit there until the heaters timed out.

scmanjarrez, 2023-04-01:
That's really weird, I fixed that problem in version 0.5.1... but I think I missed a case. I'll look into that. Thank you

tkunchick, 2023-04-01:
No worries. Pretty sure I had the same issue yesterday with my Neo v2 that I was upgrading the firmware to MRiscoC. I edit it a bit, so assumed I screwed up and just put the Creality firmware back on so I could get a couple prints done. 

If so, then its just UBL and not the standard.. Assume Creality is using Linear.

scmanjarrez, 2023-04-01:
Ok, I found the cause. I definitely forgot about the buttons when I added the new algorithm 😅

tkunchick, 2023-04-01:
Like I said, I assume user error, myself, first. when I started to see the issue on the E3V2, and remembered they both updated.. that's when I came here. All Good. :)

scmanjarrez, 2023-04-01:
Can you explain how unified ABL works? Whats the meaning of the P (phase I guess?)

tkunchick, 2023-04-01:
I don't totally understand it, pretty sure I copied and pasted from 

https://github.com/mriscoc/Ender3V2S1/wiki/Slicer-G-code-Scripts

https://marlinfw.org/docs/gcode/G029-ubl.html

P1: Automatic Probing invalidates the mesh and continues automatic probing using the probe.
In most cases the probe can’t reach all areas that the nozzle can due to the offsets specified by X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER. Deltabots can only probe within the area where DELTA_PROBEABLE_RADIUS and DELTA_PRINTABLE_RADIUS overlap.
Unreachable points can be filled in later with the P2 and P3 phases.
Use C to leave the previous mesh intact and automatically probe needed points. This allows you to invalidate parts of the mesh but still use Automatic Probing.
The X and Y parameters prioritize where to try and measure points. If omitted, the current probe position is used.
Use T (Topology) in this phase to report the probing results.
P1 will suspend mesh generation if the controller button is held down. Note that you may need to press and hold the switch for several seconds if moves are underway.

tkunchick, 2023-04-01:
I should mention, I didn't touch a 3d printer until January this year. still wrapping my head around it all. Been in IT for nearly 30 years though.

scmanjarrez, 2023-04-01:
I think I'll need the logs. I've been testing it with and without restricted ABL, but it seems to work as expected. Do you have, by any chance, the output from M420 V1?

tkunchick, 2023-04-01:
I didn't think to save it. I will get it to you in a few hours once the current print finishes.

scmanjarrez, 2023-04-01:
This version should (I hope) fix the print stopping after M420 https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.2rc0

tkunchick, 2023-04-01:
Well.. I thought I had uninstalled it, but I only disabled it, but I reinstalled, and it got the new 5.2 version. at any rate, it does work.

Is there a way I can re-install the earlier version? or are you sure you got whaat you needed already?

scmanjarrez, 2023-04-01:
I think it should "work". I haven't studied yet the unified ABL. For the time being, at least it doesn't get stuck. When you finish your print, can you give me the logs of 0.5.1? (You should be able to install the old version using the link https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.1) or using the package manager (i haven't released 0.5.2 yet, so 0.5.1 should be the one being installed)

tkunchick, 2023-04-01:
Here is the log and info from terminal. It did just stall again after getting mesh.

Wondering if its just the repeated G29 codes with different syntax.

But really, with the ABL button toggled to always, why change anything?

Also wondering, if abl was set to on, is it possible to just allow the gcode call to pass rather than replacing it with a different command?

As I understand it, 

(cut and pasted from somewhere and in my machine start code for slicer)
G29 P1    ; automatically populate mesh with all reachable points
G29 P3    ; infer the rest of the mesh values
G29 P3    ; infer the rest of the mesh values again

G29 P1 forces a new mesh..
G29 P3 forces it to fill in any gaps and make the mathematical adjustments between known points..

again, as I understand it.

[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11130707/plugin_SmartABL.log)
[abl stall.txt](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11130708/abl.stall.txt)

scmanjarrez, 2023-04-01:
I'll think about it. Also, your logs are very helpful, I now know why isn't detecting your valid/invalid mesh.

tkunchick, 2023-04-01:
Glad I could help with your work. I really like it on my Neo that still has a flat bed and rarely changes. once I get around to replacing the bed and glass (both are warped) on my E3V2, will be using it on both my machines again. save a lot of time and frustration waiting for it to do something I know is good haha.

👍 Thank you for all your work.

scmanjarrez, 2023-04-02:
Hi, can you check this version? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/0.5.2rc1 This version should take into account UBL.

piebaker, 2023-04-03:
Had the same issue, M240 S0. 0.5.2rc1 fixed it right up. Now to track down whats causing my G81.

scmanjarrez, 2023-04-03:
Can you upload plugin_SmartABL.logs? I'm taking into account G80 (for abl) and G81 (to check current mesh state) for prusa, I didn't know there was another prusa firmware that used G29.

piebaker, 2023-04-04:
> Can you upload plugin_SmartABL.logs? I'm taking into account G80 (for abl) and G81 (to check current mesh state) for prusa, I didn't know there was another prusa firmware that used G29.

[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11145493/plugin_SmartABL.log)

So a good example can be found [here.](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md) While the prusa mini is a prusa machine, it uses the usual G28/29 process as opposed to G80/G81. It's strange as this was working correctly prior to this update.

scmanjarrez, 2023-04-04:
I changed the algorithm to fix another error, but it ended having its own issues

scmanjarrez, 2023-04-04:
> [plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11145493/plugin_SmartABL.log)
>
> So a good example can be found [...]

I see, I'll release a new version adding prusa mini to the list of firmwares. Thank you!

scmanjarrez, 2023-04-05:
@piebaker do you know if prusa buddy have eeprom enabled? are you able to save mesh with M500?

piebaker, 2023-04-05:
> @piebaker do you know if prusa buddy have eeprom enabled? are you able to save mesh with M500?

No as far as I am aware M500 is disabled unfortunately. Enabling it requires physically breaking a part on the buddyboard, installing custom firmware, and voiding the warranty

scmanjarrez, 2023-04-05:
Hey @piebaker, can you check this version? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.0.0rc0 It should detect prusa-buddy.

piebaker, 2023-04-07:
> Hey @piebaker, can you check this version? [`1.0.0rc0` (release)](https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.0.0rc0) It should detect prusa-buddy.

Yup, looks like that works A-OK. Thanks so much!