scmanjarrez/OctoPrint-SmartABL #15: SmartABL continually executes ABL on every print if ABL not saved to EEPROM

AndyShap2020, 2023-04-25:
### The problem

Hi Sergio,  the add-in works really well! I've noticed an issue that may not be a bug.  When SmartABL triggers an ABL, it will run and on the actual printer screen, you have a prompt to save the newly run ABL. If you don't press save, then the print job starts. But on the next print job, SmartABL runs again. I guess this makes sense, since the ABL Mesh was not saved to EEPROM.  Is there a way for SmartABL to execute the save automatically, as opposed to me needing to manually press SAVE on the printer screen?

### Version of SmartABL

1.0.0

### Version of OctoPrint

1.87

### Operating system running OctoPrint

Version 0.18.0, running on Raspberry Pi 3 Model B Rev 1.2

### Printer model & used firmware incl. version

Ender 3v2, Mriscoc 2.13

### Browser and version of browser, operating system running browser

Firefox, Windows 10 Pro 22H2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11327076/plugin_SmartABL.log)

scmanjarrez, 2023-04-25:
Hi, thanks for reporting it. Can you share your start gcode? I suspect the problem is M420 S1 Z2; I need to confirm it and see how to fix it.

scmanjarrez, 2023-04-28:
Oh, I see the problem. This option was on early versions of SmartABL, but after the algorithm update, I removed it (unintentionally). Can you check this pre-release? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.1.0rc0

AndyShap2020, 2023-04-28:
Not quite there, it did an ABL TWICE in a row and still left the screen at Save\Continue

scmanjarrez, 2023-04-28:
What do you mean by Save/continue? I don't have anything related with that in the plugin. Can you send your logs again and a screenshot of your LCD?

AndyShap2020, 2023-04-28:
I’ll get that to you in a bit. I am at 1/5 right now.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-04-28:
Two issues:
1. As of this particular version it now runs an ABL and then immediately runs a second ABL .  
2. Didn't fix the screen popping up with Save/Continue.  While this screen is up, the printer has gone on to the next Gcode of warming up the hotend.  If it gets to temperature, it will start the print. But because I didn't click on SAVE, it will run an ABL on the next print. If I don't click SAVE on THAT print, it will run an ABL on the next print as well.
(This means that the Octoprint ABL starts between the GCode line of M190 and M109, refer to the start code above)
[image]
[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11357914/plugin_SmartABL.log)

scmanjarrez, 2023-04-29:
I don't think I can't do anything about that screen, it's from your firmware. Normal LCD screen doesn't show that dialog. Have you checked how it works? Does it pop up when you run G29?

AndyShap2020, 2023-04-29:
Yes, it does.   Not a huge deal, but you might want to make a note that for anyone using MRISCOC firmware for Ender 3v2, they need to press save, or else the ABL will not be written to EEPROM.

I am going to go back to the previous version so I can get rid of the double ABL.

THANKS!!

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-04-29:
I'll fix that, I've realized what's the problem. Does M500 work with MRISCOC firmware? I can see the M500 gcode being sent, but I don't know if the mesh is actually saved.

AndyShap2020, 2023-04-29:
I don’t know. I’ll post a question to the Facebook group.  Are you familiar with MRISCOC’s firmaware?  Jyer’s is no longer supported and MRISCOC is fantastic.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-04-29:
No... I'm currently using Klipper (also, I have LCD128240)

scmanjarrez, 2023-04-29:
This version should fix the multiple triggers: https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.1.0rc1. I'll check MRISCOC firmware if we can force save or at least avoid the viewer

AndyShap2020, 2023-04-30:
I've spoken with MRISCOC,  the G code: C108 cancels the screen waiting for confirmation.  The software DOES recognize M500, so the screen shouldn't need to be there. How do we incorporate the C108 command to run after the ABL finishes?  Would that go in the middle line for custom codes?

https://github.com/mriscoc/Ender3V2S1/wiki/Custom-C-gcodes

scmanjarrez, 2023-04-30:
I'm not really sure, try with G29, C108, M500 or G29, M500, C108

scmanjarrez, 2023-05-02:
Hey Andy, have you been able to test the latest pre-release? Does it still trigger multiple ABL? Were you able to skip mesh viewer screen?

AndyShap2020, 2023-05-02:
I think I did install it already (I’ve been running lots of prints, so not sure).  If I look at the plugin, should it say rc1, etc? Mine just says 1.1.0

I’ll reinstall after my next print. I will also add the other codes.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-05-03:
Something is wrong. I am currently on print 7/5??  It should have done an ABL the next print after 5/5.

Rc1 is installed at this point.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-05-03:
Can you upload the log file?

AndyShap2020, 2023-05-03:
I'm currently on print 9/5.  P.S. I think I added this after the install of 1.1.0rc1 as well:  Trigger Custom Codes: G29,M500,C108

[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11391416/plugin_SmartABL.log)

scmanjarrez, 2023-05-03:
Oh no, you misconfigured it. You have to put `G29, C108` in `ABL custom gcode`.
> `Trigger custom gcode` is intented for macros, or non standard default gcodes.

AndyShap2020, 2023-05-03:
DOH! I’ll put G29, M500, C108 in Custom ABL.  I’ll let you know how it goes once I get to my next print.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-05-04:
Looks like it is working fine. It DID clear the screen it DID save the mesh!

Thank you!

scmanjarrez, 2023-05-04:
Did it trigger multiple times?

AndyShap2020, 2023-05-04:
No, all appears to be working great.

commit 37a3e1a8 (2023-05-04): Added setting to enable SmartABL on unknown firmwares. Closes #15