scmanjarrez/OctoPrint-SmartABL #16: [Request] Bed Visualizer compatibility

morris2009, 2023-06-05:
### Is your feature request related to a problem? Please describe.

Currently, if SmartABL triggers a bedlevel after X days, the Bed Visualizer plugin doesn't pickup this new mesh because Bed Visualizer doesn't pick this up.

In the normal start Gcode for every print there is the code for doing a mesh, as well as a line with "@BEDLEVELVISUALIZER".
SmartABL is very good at catching the code for doing the new mesh and strip it when it isn't needed, but it leaves the @BEDLEVELVISUALIZER line.
This sets the bedlevelvisualizer in the mode that it thinks the printer is doing a bedlevel while it actually started with the print itself.
 

### Describe the solution you'd like

It would be great if SmartABL could also send a "@BEDLEVELVISUALIZER"  in the Gcode. This would have bed visualizer pick-up this new mesh and display it.

And/or if in the normal GCODE the @BEDLEVELVISUALIZER line is there and it decides no bedlevel is needed, then also strip this line from the gcode.

scmanjarrez, 2023-06-14:
Hey, any update on this? Did the suggestion work for you?

transorsmth, 2023-06-17:
This is actually a problem on the Bed level visualizer side.

scmanjarrez, 2023-06-17:
Can you explain? Anyway, you can trigger that function just by sending the command as instructed above. I checked that SABL sent the command, not that BLV triggered tho (tested on virtual printer, so I don't have access to real mesh)

transorsmth, 2023-06-17:
The commands get sent, but ABL Visualizer doesn't actually observe them unless you press the button (In the visualizer) first, which is when it sends its own gcode.

scmanjarrez, 2023-06-17:
Ohhh, I've talked with jneilliii, can you test with this steps?
1. Enable `ABL custom gcode(s)` and set it to `@BEDLEVELVISUALIZER, G29`
2. Enable `Ignore gcode(s)` and set it to `@BEDLEVELVISUALIZER`

`@BEDLEVELVISUALIZER` should be sent before `G29` in order to work

transorsmth, 2023-06-17:
I tested that, as well as just sending `@ BEDLEVELVISUALIZER` in the terminal, and neither worked (Also, to get the data back, it needs to be `G29 T`)

scmanjarrez, 2023-06-17:
I have tested it myself and works as expected. Are you using Marlin? The BEDLEVELVISUALIZER should be right after "at", i.e. `@BEDLEVELVISUALIZER`

transorsmth, 2023-06-17:
That's strange, somewhere when I was finding the visualizer gcode it said you needed the space after the @. I'm mid-print, but I'll test that as soon as its done (1h). Sorry for the trouble.

transorsmth, 2023-06-17:
It works perfectly, sorry for the trouble.

closed 2023-06-17 by scmanjarrez

morris2009, 2023-07-31:
I am sorry for the late reply.
With the suggestion of triggering on G29 and replacing it with @BEDLEVELVISUALIZER, G29 T;
it does what i want it to do, so that works!  tnx!!
Had some issues getting it to work in the beginning, not sure why, but now it does.