scmanjarrez/OctoPrint-SmartABL #17: [Request] Reset smartabl when bed leveling is run not through a file.

transorsmth, 2023-06-11:
### Is your feature request related to a problem? Please describe.

When I run a bed level visualization with Bed Visualizer, it doesn't reset the counter on smartabl. 

### Describe the solution you'd like

It would be nice if SmartABL would detect the g29 being sent to the printer, or even had a macro that would reset the counter. (In such a way that the macro could be added to the bed level visualizer level gcode)

### Describe alternatives you've considered

It would work fine if there were just a button to reset it. Possibly, if you click on the middle area that shows the count of when the next abl will be run, giving a popup to reset it.

scmanjarrez, 2023-06-15:
What do you think about a command you can send in terminal or gcode, like @SMARTABLRESET? Does it meet your request?

BCurbs, 2023-06-15:
Yes, that would be perfect!

transorsmth, 2023-06-15:
That would be perfect!

scmanjarrez, 2023-06-15:
Implemented in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.2.0. Will be available through package updater in a few hours, depending on your settings.

commit ac959fc7 (2023-06-15): Added command @SMARTABLRESET to reset counter by terminal/gcode, closes #17
closed 2023-06-15 by scmanjarrez