scmanjarrez/OctoPrint-SmartABL #18: Printer is stuck

kaenguruhs, 2023-06-24:
### The problem

1. Start the print with any of my files (example provided!) and default settings of SmartABL
2. Printer will heat the bed to 60°C, then homes all axes
3. It then heats the tool to 160° (the preheat-temperature).
4. Nothing more happens with the printer.

If I try to cancel the print, the state is stuck in "Cancelling". Deactivating the SmartABL-Plugin will solve the issue.

The start gcode is as follows:
```
; --- Global Settings
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel}

; Turn ON PSU
M80
; Extruder-Steps
M92 E460 ; Extruder steps (vorher: 445)
M900 K0.1 ; Linear-Advance for PLA

M190 S{material_bed_temperature_layer_0} ; Warten, bis vorgeheizt
G28; Fahr zum Nullpunkt
; M851 Z-2.52; Nozzle to probe offset PETG
; M851 Z-3.17 ; PLA
M104 S160 ; extruder auf 160 ° heizen
@BEDLEVELVISUALIZER
G29; Autoleveling
M420 S1; Get and/or set bed leveling state
M109 S{material_print_temperature_layer_0} ; extruder korrekt heizen
G92 E0; setze den Extruder auf Null
G1 Z2.0 F3000; Druckkopf 2mm nach oben fahren
G1 X10.1 Y20 Z{layer_height_0} F5000.0; Fahre an die Startposition X10.1 Y20 Z0.28
G1 X10.1 Y200.0 Z{layer_height_0} F1500.0 E25; Ziehe eine Linie von der Startposition aus nach Y200
G1 X10.4 Y200.0 Z{layer_height_0} F5000.0; Bissl zur Seite bewegen
G1 X10.4 Y20 Z{layer_height_0} F1500.0 E50; eine 2. Linie nach vorne ziehen
G92 E0; setze den Extruder auf Null
G1 Z2.0 F3000; Fahr die Düse 2mm nach oben
;M201 X2000.00 Y500.00 Z100.00 E10000.00
```

The last line I can see in the terminal is `M104 S160`:
```
Changing monitoring state from "Operational" to "Starting"
Send: N4 M117 ETA is 15:35:03*75
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Recv:  T:171.27 /0.00 B:57.37 /0.00 @:0 B@:0
Send: N1 M82*24
Recv: ok
PSUControl: ok
Send: N2 M92 E460*77
[...]
Recv: ok
Send: N3 M117 ETA is 15:32:27*77
Recv: ok
Send: N4 M117 ETA is 15:32:28*69
Recv: ok
Send: N5 M117 ETA is 15:32:12*77
Recv: ok
Send: N6 M900 K0.1*104
Recv: ok
Send: N7 M190 S60*89
[...]
Recv: ok
Send: N8 M113 S2*105
Recv: ok
Send: N9 M117 ETA is 15:32:04*70
Recv: ok
Send: N10 M117 ETA is 15:32:14*127
Recv: ok
Send: N11 M117 ETA is 15:32:24*125
Recv: ok
Send: N12 G28*32
[...]
Recv: X:128.00 Y:187.00 Z:17.78 E:0.00 Count X:10255 Y:14983 Z:7108
Recv: ok
Send: N13 M117 ETA is 15:32:31*123
Recv: ok
Send: N14 M117 ETA is 15:32:41*123
Recv:  T:141.09 /0.00 B:62.13 /60.00 @:0 B@:0
Recv: ok
Send: N15 M104 S160*86
Recv: ok
Send: N16 M420 V1*85
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.117 +0.069 +0.054 +0.054 +0.127
Recv:  1 +0.094 +0.037 +0.019 +0.014 +0.029
Recv:  2 +0.129 +0.092 +0.054 +0.042 +0.022
Recv:  3 +0.049 -0.021 -0.026 +0.002 -0.041
Recv:  4 -0.016 -0.066 -0.068 -0.046 -0.001
Recv: 
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok
Send: N17 G29*36
[...]
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.069 +0.027 +0.019 -0.003 +0.054
Recv:  1 +0.049 +0.012 -0.023 -0.021 -0.013
Recv:  2 +0.072 +0.044 +0.017 -0.018 -0.028
Recv:  3 -0.018 -0.053 -0.066 -0.043 -0.063
Recv:  4 -0.104 -0.129 -0.134 -0.099 -0.056
Recv: 
Recv: echo:busy: processing
Recv:  T:159.63 /160.00 B:60.03 /60.00 @:23 B@:0
Recv: X:258.00 Y:310.00 Z:15.00 E:0.00 Count X:20671 Y:24838 Z:5967
Recv: ok
Send: N18 M117 ETA is 15:32:32*115
Recv: ok
Send: N19 M117 ETA is 15:32:42*117
Recv: ok
Send: N20 M117 ETA is 15:32:52*126
Recv: ok
Send: N21 M117 ETA is 15:33:02*123
Recv: ok
Send: N22 M117 ETA is 15:33:12*121
Recv: ok
Send: N23 M117 ETA is 15:33:22*123
Recv: ok
Send: N24 M117 ETA is 15:33:32*125
Recv: ok
Send: N25 M500*17
Recv: echo:Settings Stored (709 bytes; crc 57400)
Recv: ok
Send: N26 M420 V1*86
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.069 +0.027 +0.019 -0.003 +0.054
Recv:  1 +0.049 +0.012 -0.023 -0.021 -0.013
Recv:  2 +0.072 +0.044 +0.017 -0.018 -0.028
Recv:  3 -0.018 -0.053 -0.066 -0.043 -0.063
Recv:  4 -0.104 -0.129 -0.134 -0.099 -0.056
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok
Recv:  T:160.13 /160.00 B:59.69 /60.00 @:16 B@:11
Send: N27 M117 ETA is 15:33:37*123
Recv: ok
Recv:  T:160.35 /160.00 B:59.71 /60.00 @:14 B@:8
Recv:  T:160.13 /160.00 B:59.84 /60.00 @:16 B@:1
Recv:  T:159.85 /160.00 B:59.79 /60.00 @:20 B@:6
Recv:  T:159.96 /160.00 B:59.80 /60.00 @:19 B@:7
Recv:  T:159.67 /160.00 B:59.38 /60.00 @:22 B@:29
Send: N28 M117 ETA is 15:33:47*115
Recv: ok
[...]
```

### Version of SmartABL

1.2.0

### Version of OctoPrint

1.9.0

### Operating system running OctoPrint

RaspiOS with docker

### Printer model & used firmware incl. version

Sidewinder X1

### Browser and version of browser, operating system running browser

-

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230624150405.zip](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11857033/octoprint-systeminfo-20230624150405.zip)
[plugin_SmartABL (2).log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11857036/plugin_SmartABL.2.log)
[PipeHolder.aw.gcode.txt](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11857037/PipeHolder.aw.gcode.txt)

scmanjarrez, 2023-06-24:
Ok, I found the bug. Can you test removing M420 S1 from your gcode?

scmanjarrez, 2023-06-24:
Can you test this pre-release? https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.2.1. It should fix the pause problem

kaenguruhs, 2023-06-24:
Yep, this 1.2.1 pre-release version works :-)
I tested with the same settings & same files mentioned. Thanks a lot!

scmanjarrez, 2023-06-24:
Fixes in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.2.1

closed 2023-06-24 by scmanjarrez