scmanjarrez/OctoPrint-SmartABL #19: Still shows version 1.2.0 after updating to 1.2.1

smhawkes, 2023-06-25:
### The problem

1. update to 1.2.1
2. restart Octoprint
3. pop shows that an upgrade to 1.2.1 is available

### Version of SmartABL

1.2.1

### Version of OctoPrint

1.9.0

### Operating system running OctoPrint

OctoPi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230625150451.zip](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/11861006/octoprint-systeminfo-20230625150451.zip)

scmanjarrez, 2023-06-25:
Ups, sorry, I completely missed the version update!

scmanjarrez, 2023-06-25:
Fixed in https://github.com/scmanjarrez/OctoPrint-SmartABL/releases/tag/1.2.2

closed 2023-06-25 by scmanjarrez