scmanjarrez/OctoPrint-SmartABL #21: SMartABL not working after update to Octoprint 1.9.1

AndyShap2020, 2023-07-18:
### The problem

Nothing changed other than update to Octoprint 1.9.0, then .1 then .2.
No ABL after print 5 (I have it set to 5 prints/ 10 days.

### Version of SmartABL

1.2.2

### Version of OctoPrint

1.92

### Operating system running OctoPrint

OctoPi 1.0.0

### Printer model & used firmware incl. version

E3V2, MRISCOC Ender3V2S1-20221002

### Browser and version of browser, operating system running browser

Firefox, latest

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[plugin_SmartABL.log](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/12087880/plugin_SmartABL.log)
[octoprint-systeminfo-20230718111159.zip](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/12087884/octoprint-systeminfo-20230718111159.zip)

Start and end codes unchanged since last issue reported.

AndyShap2020, 2023-07-19:
> I don't see any problem. The counter hasn't reached 5 prints, so it doesn't trigger. It only triggered 1 time 2023-07-18 10:51:23,398 when the counter reached 5 prints.

No something is wrong.  I watched it go from 4/5 to 5/5 to 1/5 and no ABL was triggered.

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

scmanjarrez, 2023-07-19:
I don't know what to say, the logs say it sent G29, M500, C108...

AndyShap2020, 2023-07-19:
Let me do a little more home work.  I’ll generate a log when it finishes print 5/5 and then I’ll generate the log when it finishes print 1/5

AndyShap2020, 2023-07-20:
I watched it carefully this time.  When I started print 1/5 (screen reads 0/5) it hovered over the center position, waited a moment and then the EV32 screen beeped with “settings saved”, it then started the print. So, no ABL occurred, but I think it THOUGHT it occurred.  I have C108 added to my codes, I removed that so that I could see further what the printer screen was reporting.

scmanjarrez, 2023-07-20:
That's really weird, can you test with the "ABL Always" enabled?

AndyShap2020, 2023-07-20:
Yes. I am starting in on some longer prints, so it may be a few days for good results.

AndyShap2020, 2023-07-20:
No, something is wrong. :-(
Screen print showing ABL Always is on: 
[image]

Video showing that no ABL is happening. Look at the printer screen at the :12 second mark. You will see that the printer screen says "settings stored" as if it did an ABL.

Also, I removed the C108 command, so it should be showing the ABL screen results.

https://github.com/scmanjarrez/OctoPrint-SmartABL/assets/71109919/38bddff2-3fc5-41e1-a8d2-8b559466adc4

[octoprint-systeminfo-20230720131708.zip](https://github.com/scmanjarrez/OctoPrint-SmartABL/files/12113792/octoprint-systeminfo-20230720131708.zip)

scmanjarrez, 2023-07-20:
Can you send G29 from octoprint?

AndyShap2020, 2023-07-20:
Well, you are probably on to something. Nothing happens when I send the command:
Send: G29
Recv: ok P15 B15
Recv:  T:69.02 /0.00 B:53.09 /0.00 @:0 B@:0
Recv:  T:68.13 /0.00 B:53.03 /0.00 @:0 B@:0
Recv:  T:67.34 /0.00 B:52.96 /0.00 @:0 B@:0
Recv:  T:66.52 /0.00 B:52.88 /0.00 @:0 B@:0
Recv:  T:65.78 /0.00 B:52.83 /0.00 @:0 B@:0

scmanjarrez, 2023-07-20:
Have you updated your firmware? Have you tried disabling SmartABL and sending G29?

AndyShap2020, 2023-07-20:
SmartABL: is disabled and still G29 does not work. I guess you are off the hook and I’ll have to figure out why it stopped working

(G28) works fine)

Mriscoc software is newest version.

closed 2023-07-26 by scmanjarrez